balena-io/etcher #1116: Identity partition table type in analytics & error reporting

jviotti, 2017-02-19:
I'd be great if we can know in advance if an image has a MBR or a GPT.

lurch, 2017-03-02:
If we're gathering this info for analytics purposes, it might also be useful to collect which filesystems are being used in the images being written? (FAT16/FAT32/exFAT/ext3/ext4/btrfs/etc.)

I'll assign this to @jhermsmeier because he already has modules to do some of this, but I dunno what priority this task has.

jhermsmeier, 2017-11-29:
Closed in https://github.com/resin-io/etcher/pull/1248

closed 2017-11-29 by jhermsmeier

lurch, 2017-11-29:
Apologies if I'm being dense, but what part of #1248 closes this? All I can see is
`analytics.logEvent('Missing partition table', image);`

jhermsmeier, 2017-11-29:
The image metadata gets logged, which is what contains the MBR/GPT info

lurch, 2017-11-29:
Ah, thanks. Does this also include the partition / filesystem types, as I mentioned in my comment above?

jhermsmeier, 2017-11-29:
Yup it does