balena-io/etcher #1124: Improve ENOENT/ENXIO errors when unplugging drive while flashing/validating, and prevent them from reaching TrackJS

jviotti, 2017-02-21:
### Unplugging while flashing

[image]

[image]

### Unplugging while validating

[image]

[image]

- [ ] Add and document common error codes for these errors in [`etcher-image-write`](https://github.com/resin-io-modules/etcher-image-write)
- [ ] Handle such codes in the GUI, preventing them from getting to TrackJS

jviotti, 2017-02-22:
As a bonus point (very, very low priority), we could allow a "resume" option for the writer to continue from where it stopped if the drive is plugged back.

lurch, 2017-02-23:
As I mentioned in gitter, I don't think that's actually possible, since if the drive was removed unexpectedly AFAIK there's no guarantee about the order that the blocks actually got written by the FlashTranslationLayer? i.e. we might have "written" blocks 0-50 to the card, but perhaps only blocks 0-22 and 34-41 actually made it out of the write buffers and into the card's physical flash memory?

alexandrosm, 2017-02-23:
In that case we can enter a fast-forward "matching" mode, where we read
what's on the card, and if it matches what we're about to write, we move on
to the next, until we find a disagreement, at which point we continue with
a regular write.

Having said all this, this sounds like a lot of work for a tiny
improvement, so we really should not do this unless we're scraping the
bottom of the barrel for optimisations.

jviotti, 2017-04-15:
This issue was fixed by https://github.com/resin-io/etcher/pull/1301

closed 2017-04-15 by jviotti