balena-io/etcher #1126: Fix "Command Failed" error when unplugging a drive exactly when the drive scanning script runs

jviotti, 2017-02-21:
Steps to reproduce:

- Unplug a drive *exactly* when the drive scanning script is running

You might want to add `console.log` above https://github.com/resin-io/etcher/blob/master/lib/gui/modules/drive-scanner.js#L41 to know when exactly the script is running,  or decrease the drive scanning interval to increase the chances: https://github.com/resin-io/etcher/blob/master/lib/gui/modules/drive-scanner.js#L36

This is the error I get:

[image]

jviotti, 2017-02-21:
I confirm this issue only happens in OS X. I can't reproduce on GNU/Linux nor Windows.

lurch, 2017-02-22:
This is probably highly timing-dependent, so maybe it's just harder to reproduce on other platforms?
All the drivelist platform-scripts effectively do:
```
drives = list_drives()
foreach drive in drives:
    get_more_info(drive)
```
so it's conceivable that _any_ of the scripts would fail if a drive disappears between `list_drives` and `get_more_info`.
In fact I just tried on Linux `lsblk -b -d "/dev/sdd" --output SIZE,RO,RM,MODEL` (I don't have any `/dev/sdd` currently) and it errored with `lsblk: /dev/sdd: not a block device` and set exit-code 1.

Maybe this is something that should be handled at the `drivelist` level (i.e. making all the scripts robust to drives disappearing between `list_drives` and `get_more_info`), rather than letting the error bubble up to Etcher itself?
(and of course if there are multiple steps in the `get_more_info` process, then the drive could potentially disappear between any of those steps - which I guess means we'd need to internally buffer the info (as we're gathering it for each drive), rather than just printing directly to STDOUT)

jviotti, 2017-02-22:
Yeah, maybe. I've tried reproducing this for a good while without success, but you might be right in that it may be just harder to reproduce in other platforms.

jviotti, 2017-02-22:
> Maybe this is something that should be handled at the drivelist level (i.e. making all the scripts robust to drives disappearing between list_drives and get_more_info), rather than letting the error bubble up to Etcher itself?

Sounds like the way to go!

lurch, 2017-02-23:
>  I've tried reproducing this for a good while without success

I guess a easy way to 'force' that would be to just put a big `sleep 10` between the `list_drives` and `get_more_info` steps ;-)

jviotti, 2017-02-23:
That's a good idea. Turns out there were errors on Windows:

```
C:\cygwin64\home\jviotti\Projects\etcher\node_modules\drivelist\scripts\win32.bat?.wsf(79, 3) (null): 0x80041002
```

However the script exits with code 0, and since the output is invalid YAML, it gets only printed to DevTools instead of showing an error alert, which explains why I've not been seeing it.

The weird thing is that if I execute the script directly (from `cmd.exe`), then it exists with code 1, so `drivelist` should be catching this :/

jviotti, 2017-02-23:
OK, so we were not passing the VBScript exit code to the caller from within the `.bat` wrapper. I'll send a PR to `drivelist` in a minute. This is what I get after my changes:

[image]

jviotti, 2017-02-23:
See https://github.com/resin-io-modules/drivelist/pull/146

jviotti, 2017-02-23:
Adding a timeout, as @lurch suggested, revealed the following on Linux:

[image]

closed 2017-03-06 by jviotti