balena-io/etcher #1127: Improve error when unplugging the drive while its getting erased on Windows

jviotti, 2017-02-21:
Steps to reproduce:

- Select image
- Select drive
- Click "Flash"
- Unplug the drive while the button says "Starting..."

This is the error I get:

[image]

This seems to be the same as https://github.com/resin-io/etcher/issues/989

We should probably detect if the drive exists before attempting to retry the `diskpart` erasure procedure, and if so, fail with another error that should not get to TrackJS.

jviotti, 2017-05-09:
This was already addressed by https://github.com/resin-io-modules/etcher-image-write/commit/e2d3f0d1b7f62054c34835a731c38a87b6c1fa29

closed 2017-05-09 by jviotti

lurch, 2017-05-10:
Minor nit, but IMHO it'd be nicer to link to the PR ( https://github.com/resin-io-modules/etcher-image-write/pull/99 ) rather than the commit?

jviotti, 2017-05-10:
True, sorry about that!