balena-io/etcher #1136: Display the determined uncompressed size on the "too small for image" warning

jviotti, 2017-02-25:
We should make sure the under understands why the drive is not large enough, which can be a bit confusing in the case of compressed images, where the compressed package is usually smaller than the drive.

Shou, 2017-02-27:
Something like this? Although maybe "image size too big" "actual image size too big" or some such?

[image]

lurch, 2017-02-27:
Might also be worth having an option to display the exact byte-sizes of both the uncompressed image and the drive in that scenario too, since with the units currently being always-rounded to GB, a "7.5GB" image may not actually fit on a "7.5GB" drive, if the image is even a few kilobytes larger than the drive.

lurch, 2017-02-27:
Hmm, looking at @Shou 's screenshot I guess we need to be careful to try and keep the "warning label" small - maybe @konmouz has some ideas?

Shou, 2017-02-27:
@lurch yeah, good call. Perhaps "Image Size Bigger By x units" and we just display in the appropriate human readable unit?

jviotti, 2017-02-27:
Let's bring @konmouz here.

konmouz, 2017-02-27:
2 options:
- we can have a 'insufficient space' label and on hover a tooltip with details (what @lurch suggested)
- avoid the label and have just an the alert/ danger icon with the explanation, all in red - no tooltip

BTW, @Shou why the above element has two labels? System and boot? It would be better to stick to 1 label per drive.

jviotti, 2017-02-27:
I think I like the second option more, given that a tooltip might not be very discoverable unless the user however on the badge. In order to continue the brainstorm session: what about displaying the size that doesn't fit instead? E.g: You need 300 more MBs for the image.

konmouz, 2017-02-27:
@jviotti this sounds even better. So using the label font size and the alert icon, we can display such a message.  - Or, 300 MBs more are required for the selected image. -

lurch, 2017-02-27:
> BTW, @Shou why the above element has two labels? System and boot?

It's actually System and Source ;-)  But yeah, it probably makes sense to only show one warning-label-per-drive, and I guess in this instance `Source` is 'more important' than `System` (a system drive *may* be writable, but a source drive (where the image is being read from) never will be).

With regards to the "too small" label message, what about "Drive is 300MB smaller than image" ?

konmouz, 2017-02-27:
Sounds good to me, we should probably add the word 'selected' though, just to help user understand.

alexandrosm, 2017-02-28:
The "source" label doesn't sound like something I'd understand, nevermind
my mother. Shouldn't we find a better name for it?

lurch, 2017-02-28:
I guess `Source` was chosen, as it's the logical-opposite of `Destination`, but you're right that it doesn't make much sense when seen in isolation. Difficult to thing of a more meaningful label though... what about something like "Image source" or "Image location" instead of "Source Drive"?

jviotti, 2017-02-28:
I created new issues to track some of the suggestions mentioned here: 

- https://github.com/resin-io/etcher/issues/1143
- https://github.com/resin-io/etcher/issues/1144

commit ad7c8760 (2018-03-21): Merge pull request #1807 from resin-io/patch-1136

feat(gui): Display image size when drive too small

Close #1136
Close #2004
closed 2018-03-21 by jhermsmeier