balena-io/etcher #1137: After burning, when ejecting, get a command failed.

n1rvana, 2017-02-26:
- **Etcher version:**
v1.0.0-beta.19 - 2017-02-24

- **Operating system and architecture:**
MacOS Sierra MacbookPro Touchbar

- **Reproduction**

- load iso, burn iso
- When complete, hit mac brings up dialog about unreadable drive
- hit eject button on that dialog

get dialog that says:

Dialog pops up with this:
Error: Command failed: /var/folders/0m/p1ln6f7n4015188cb260x4gm0000gn/T/.io.resin.etcher.wMwwzj

    at ChildProcess.exithandler (child_process.js:218:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

- **Do you see any meaningful error information on DevTools?**

DevTools won't  let me copy and paste ,but prints the stack trace above plus:
"a.(anonymous function) @ tracker.is:10

jviotti, 2017-02-27:
Thanks a lot for reporting! Can you link me to the exact image you're flashing, so I can ensure I can reproduce?

lurch, 2017-02-27:
Just a complete guess, but maybe the Mac OS itself is complaining about unreadable drive, and it's the Mac OS that is "ejecting" the unreadable drive, and so Etcher is erroring when it's trying to unmount an already-ejected drive? i.e. maybe this is a variation of #1128 ?

jviotti, 2017-02-27:
Yeah, I'm thinking the same. In any case, we only release the file descriptor after validation, right before unmounting, so you would have to be really fast to confirm the dialog for that to happen. I still can't reproduce, it gets unmounted before the OS is able to realize the file system is unknown.

lurch, 2017-05-03:
Perhaps this has been fixed by the native mountutils module?
@n1rvana Could you please check to see if the latest version of Etcher still has this problem?

jhermsmeier, 2018-03-09:
Closing this as there's no response or new instances of this, and this has most likely already been fixed by updates to either drivelist or mountutils.

closed 2018-03-09 by jhermsmeier