balena-io/etcher #1138: Slow start

PrzemekSkw, 2017-02-27:
- **Etcher version: Etcher-1.0.0-beta.19**
- **Operating system and architecture: Fedora 25 -Gnome, 64 bit**
- **Do you see any meaningful error information on DevTools? No**

Hi, it's possible to improve launch speed of Etcher because it's slowest app I have used. I have fast SSD drive and every app starting when I click on icon but for Etcher I must wait about 6 seconds :)

Regards,

lurch, 2017-02-27:
I believe this is a side-effect of AppImages and/or the executable-compression used by Etcher.
Changing either of those would make Etcher much harder to install (i.e. we wouldn't be able to use a single download to support all Linux distros), and make it take up lots more diskspace / be an even bigger download.
Unless @probonopd has some suggestions?

PrzemekSkw, 2017-02-27:
OK, thanks for an answer. Maybe flatpak will be better option than appimage ?

jviotti, 2017-02-27:
Its probably a combination of:

- AppImages
- Electron (which is not particularly a "fast" framework)
- UPX

I'll prepare some custom builds so we can try to find the current bottleneck.

lurch, 2017-02-27:
Coincidentally, I had a quick look at the flatpak website over the weekend, but unfortunately it doesn't support Ubuntu 14.04 (which I'm still running on my laptop), so I didn't look at it in much depth...

probonopd, 2017-02-27:
UPX? Why UPX? You could try squashfs-based AppImages for comparison, too. And the unpacked AppImage (then: AppDir).

jviotti, 2017-02-27:
UPX seems to slow down the binary considerably in some cases when we use the best possible compression. I'm uploading a set of builds as we talk, which should be enough to at least identity what to investigate further.

probonopd, 2017-02-27:
AppImages are compressed without the need for UPX anyway...

PrzemekSkw, 2017-02-27:
I try Kaku Music Player appimage and it start in 1 second :)

jviotti, 2017-02-27:
Yeah, I doubt AppImages are the cause. I'm still uploading btw, I should probably move to a country with better internet :P

jviotti, 2017-02-27:
@PrzemekSkw Here are the builds:

- AppImage without upx compression: https://drive.google.com/open?id=0B7tkbonGU-RyWEN1bGJqT0tWRDQ
- Raw build (without being packaged as an AppImage) without upx compression: https://drive.google.com/open?id=0B7tkbonGU-RyOGVTM1prX0wwVDA
- Raw build with upx compression: https://drive.google.com/open?id=0B7tkbonGU-RySlA2bS1jT0VKYlk

Let me know how that goes!

Sobient, 2017-03-17:
I've noticed this too, and it's so apparent that it made me wonder if something was wrong. While 6-7 seconds isn't much, it's very noticeable when most of that time nothing is happening. As a matter of fact it's the one of the first programs I've ever encountered to not "launch right away".

jviotti, 2017-03-17:
Hi @Sobient 

> While 6-7 seconds isn't much, it's very noticeable when most of that time nothing is happening

6-7 seconds its definitely not normal. What are the specs of your PC? I'm on a MacBook Pro with an i5 and it takes 2 seconds here.

> @jviotti : Any chance you could build windows-versions without the compression?

Sure, will do, probably later today or earlier next week. I'll let you know.

PrzemekSkw, 2017-03-17:
Thanks very much, that second Raw build works faster :)

jviotti, 2017-03-17:
@PrzemekSkw ,

Thanks a lot for the response. Between the other two, which one runs faster?

PrzemekSkw, 2017-03-17:
Both Raw builds start in 3 seconds :)

jviotti, 2017-03-17:
So AppImages seem to be the bottleneck here?

PrzemekSkw, 2017-03-17:
@jviotti don't know what is means (bottleneck) but appimages are slower :)

jviotti, 2017-03-17:
Regarding GNU/Linux, I think we should try to upgrade to the next generation of AppImages, and see if that improves things. 

I'll also build Windows images and we'll see how that goes.

tooomm, 2017-03-17:
> 6-7 seconds its definitely not normal. What are the specs of your PC? I'm on a MacBook Pro with an i5 and it takes 2 seconds here.

Totally normal (and unacceptable) on Windows... 👎 
I upgraded to beta19 64bit (portable) just to test again: 7 seconds for me on 64bit Win10 (i5-3320M), SSD

probonopd, 2017-03-17:
> Regarding GNU/Linux, I think we should try to upgrade to the next generation of AppImages, and see if that improves things.

...and remove upx :-)

lurch, 2017-03-17:
Just to make it not look like "nothing is happening", would there be any way to (immediately) display a loading "splash screen" like some applications use?

jhermsmeier, 2017-03-17:
Not until upx is done unpacking, I believe

jviotti, 2017-03-17:
I agree. Let's kill upx. The savings the tool provide are not worth it if it slows down the application. We can re-gain those MBs by concatenating and minifying JavaScript, ignoring unused code.

Sobient, 2017-03-18:
@jviotti : I currently run an overclocked i7 k3770 and tons of RAM, so that shouldn't be the issue. As said, no other program I start has this issue (for example: Virtualbox, Libre Office, Screamer Radio, NESTopia)

JaredZena, 2017-10-06:
So if Etcher just got freezed and does nothing else is it ok to restart the whole process again? Can it damage my SD card if I do that?

probonopd, 2017-10-09:
> I agree. Let's kill upx. The savings the tool provide are not worth it if it slows down the application. We can re-gain those MBs by concatenating and minifying JavaScript, ignoring unused code.

...or someone could volunteer to implement zstd support for AppImage ,-)
https://github.com/AppImage/AppImageKit/issues/478

tooomm, 2018-01-15:
The bigger GitHub Desktop app is also Electron based, and starts up considerably faster, maybe they handle things different somehow?
https://github.com/desktop/desktop

The portable app is unusable by the way.
I had to check the task manager after 20s of "nothing", to see if my double click got recognized at all. 
Startup overall was 40s. :(

probonopd, 2018-01-15:
`etcher-1.2.1-x86_64.AppImage` has 3 seconds "cold" startup time (never opened before on this system) on a moderate system (C2D E8400, 8 GB RAM, running Xubuntu 16.04 from a Live ISO stored on SSD).

jhermsmeier, 2018-01-15:
I think @tooomm is talking about the portable Etcher on Windows; but still 40 seconds seems excessive. We'll have a look at what might be going wrong there.

tooomm, 2018-01-15:
Exactly, I was running `Etcher-Portable-1.2.1-x64.exe` with Windows 10 64bit.
On a i5-3320M + 16GB + SSD by the way @probonopd.

To provide more data, I explicitly installed `Etcher-Setup-1.2.1-x64.exe` as well.
It takes 11s to start in comparison - still slow.
Why does Etcher perform that bad on Windows?

lurch, 2018-01-15:
Sometimes the blame is anti-malware apps (e.g. `MsMpEng.exe`) scanning the Etcher executable (at least, this seems to be the primary cause of slowdown on my old Windows 7 laptop). Are your definitions up to date? Could you open up Task Manager before launching Etcher portable, to see if you can identify which process is hogging all your PC's resources?

tooomm, 2018-01-15:
>Are your definitions up to date? 

Windows has all updates installed. No additional Anti-Malware is used (besides Windows Defender).

>Could you open up Task Manager before launching Etcher portable, to see if you can identify which process is hogging all your PC's resources?

For the portable app, the system is bored during all startup...
CPU is at 20-25% @1.2GHz (lowest possible clock), and reaching 35% @1,4GHz a few seconds before it shows up. Turbo would go up to 3.3GHz I think.
SSD is at 2-6% load all the time. (RAM usage stable at 43%)

lurch, 2018-01-15:
https://ark.intel.com/products/64896/Intel-Core-i5-3320M-Processor-3M-Cache-up-to-3_30-GHz says that the i5-3320M is a 2-core, 4-thread CPU (i.e. 4 virtual cores). I wonder if it's possible that the 25% figure you're seeing is simply a single thread maxing out at 100%? (I don't have Windows 10 myself, so I dunno how it reports CPU usage). Or is it only 25% of only a single virtual core? Is Etcher.exe the only process that's using up any CPU-time (while it loads)?

tooomm, 2018-01-15:
>I wonder if it's possible that the 25% figure you're seeing is simply a single thread maxing out at 100%? 

If that would be the case, shouldn't the CPU clock increase?

>Is Etcher.exe the only process that's using up any CPU-time (while it loads)?

No, CPU load before is ~6-8% (mostly Firefox with several windows+tabs).
Etcher consumes during start up an additional ~5-15%.
I also realize that Etcher splits up into 4 "Etcher" sub-processes
and 1 "Flash OS images to SD cards and USB drives..." process.

closed 2018-10-15 by jviotti

tooomm, 2018-10-15:
@jviotti what changed here?

nebrelbug, 2019-10-29:
I'm running AppImage on Pop!_OS (based on Ubuntu). Takes significantly longer than my other applications to open, and longer than it did on Windows. Any updates?

probonopd, 2019-10-30:
> Takes significantly longer

Can you please give some measurements in seconds?

What happens if you extract the AppImage with `--appimage-extract` and run the `AppRun` file contained therein?