balena-io/etcher #1139: Error when closing the new version notification popup

emirotin, 2017-02-27:
- **Etcher version:** beta.18
- **Operating system and architecture:** macOS 64bit
- **Do you see any meaningful error information on DevTools?**

[image]

lurch, 2017-02-27:
I've just checked, and I see the same in Linux :-/

jviotti, 2017-02-27:
At which stage is this happening? Right when you open the app?

lurch, 2017-02-27:
Start version beta18 of the app (I'm kinda 'cheating' by running `/usr/share/etcher-electron/etcher` from the terminal, which skips setting `ETCHER_DISABLE_UPDATES` (I haven't `apt-get upgrade`d yet)), and the "NEW UPDATE AVAILABLE!" dialog appears. Click on the "X" or on "SKIP", and the error shown by Eugene then appears :-/
(Clicking 'OK' on that error dialog, and Etcher seems to continue to run fine as normal)

jviotti, 2017-02-27:
I can confirm this issue was fixed by https://github.com/resin-io/etcher/pull/1083!

closed 2017-02-27 by jviotti

lekkas, 2017-03-01:
Just bumped into this after selecting `skip` in the update notification. Using `beta18`, I'll update to the latest version

jviotti, 2017-03-01:
Yeah, this bug is trending massively high on TrackJS as you can imagine :P