balena-io/etcher #1142: Error: spawn UNKNOWN on Windows during burn

mkeegan, 2017-02-27:
Hi folks! Mathew over at [Kano](https://www.kano.me) here. We love Etcher, and now recommend it to all 100,000 of our customers!

We've noticed a few folks on Windows running into this rather cryptic error message. They're using v1.0.0-beta.19 trying to install Kano OS 3.8.0. 

[image]

We're pretty exclusively Mac-based at HQ, but we tried a Windows 10 machine and couldn't replicate the issue. Is there any more info you'd like us to follow up with?

jviotti, 2017-02-27:
Hi Mathew,

Thanks a lot for reaching out :) We're always chasing these obscure Windows issues, which seem to always happen sporadically.

Are you sure this is on v1.0.0-beta.19? The screenshot seems to be running v1.0.0-beta.17.

It'd be nice to know at which stage this error happened. E.g: did it happen while flashing? If so, where? during the flash stage? During the validation stage? At which point? At the beginning of the validation phase? 

Such information would help us sketch some hypothesis of how this could happen, so we can have a better chance of reliably reproducing.

Here are some thoughts based on the information we have at the moment:

The error is during spawn (the `execFile` function, precisely). There are a couple of places this happens:

- The drive detection system

At a certain interval (2 seconds at the moment), we spawn a VBScript script that communicates with WMI to get the current drives information. This starts happening as soon as the app is started, but stops during flashing.

- The elevation system

In order for the elevation mechanism to take place, two spawns happen in the middle: one to re-execute the child writer proxy with elevation once the elevation dialog is accepted, and another one when the writer process is started.

- Drive unmounting

We unmount the drive before and after the flash.

***

Another good information to have is the exact image that was used to flash the error, and also the full path where the image was located (path weirdness could be the issue, so we should discard that as well), and whether Etcher was installed to program files, or was ran using the standalone ZIP that we publish on the website.

Thanks a lot again, we'll hopefully get to the bottom of it!

jviotti, 2017-02-27:
I'm leaving out the "component:" tag since we don't know where exactly the issue came from.

jviotti, 2017-02-27:
Also: do you know if the user can actively reproduce this? If so, could you ask him to open DevTools (with Ctrl+Alt+I) and paste the whole thing here?

mkeegan, 2017-02-28:
Looks like copy/paste fart on the version number. Thanks for the extra info, I'll create a bulletin sheet for the customer service folks to get more info when these happen!

jviotti, 2017-02-28:
Sounds great, let us know how that goes!

jviotti, 2017-04-15:
I accidentally found the root cause of this issue. It happens when you unplug an SD Card during flash/validation from an internal reader (related to https://github.com/resin-io/etcher/issues/1124).

I'll provide a nice user friendly error message like in the other cases.

jviotti, 2017-04-15:
[image]
[image]

jviotti, 2017-04-15:
Oh, wait. The original error says `spawn UNKNOWN`, while this says `UNKNOWN: unknown error`. Maybe its yet a different one?

jviotti, 2017-05-09:
This is definitely coming from drivelist. Sentry shows similar errors on 1.0.0-rc.5, in which we don't spawn for elevation nor for unmounting anymore

closed 2017-07-12 by jviotti

lurch, 2017-07-13:
Please see https://github.com/resin-io/etcher/pull/1574#issuecomment-315054168

PR #1574 (2017-07-12): upgrade: drivelist to v5.0.25
This version contains various improvements on the drive detection system
mainly on Windows. The improvements can be summarized as follows:

- Drivelist no longer spawns a .bat script to perform its job. This
  means that a lot of spawning related issues are now fixed

- Drivelist doesn't fetch drive sizes from WMI anymore, which was known
  to incorrectly report a very small size for certain drives, causing
  the famous "0.0 GB" Windows issues (see
  resin-io-modules/drivelist#142)

- Cleanup temporary scripts after execution

This PR also sets the DRIVELIST_DEBUG [... cut]