balena-io/etcher #1143: Don't show System Drive and Source Drive badges at the same time

jviotti, 2017-02-28:
From https://github.com/resin-io/etcher/issues/1136

[image]

Which one should we prefer though? @konmouz ?

WasabiFan, 2017-02-28:
I think there's a case to be made for keeping both: Doing so reinforces the meaning of "source drive" (or whatever you rename it to) so that the user has a reasonable means of understanding that label.

jviotti, 2017-02-28:
That's a good point as well. Up to @konmouz I guess.

konmouz, 2017-02-28:
@jviotti I am not sure if I am the better to answer this, it is a bit technical. Is source drive a boot drive? We need to find a name that represents this drive and fits in one label.

jviotti, 2017-02-28:
@konmouz The source drive is the drive that contains the image that you previously selected (not necessarily a system drive)

konmouz, 2017-02-28:
@jviotti, ok I was completely out of topic. Now it makes sense why you might need two. So, since a 'source' drive is only one at a time, it should have a different color label (for instance orange). Also, may worth renaming to 'image source' and of course the drive would be greyed out. Maybe in that sense, we don't need the second label even if the drive is system or boot drive.

jviotti, 2017-02-28:
Perfect, I'll close this issue then!

lurch, 2017-02-28:
> Now it makes sense why you might need two. (...)
Maybe (...) we don't need the second label even if the drive is system or boot drive.

I'm not quite sure I follow. Which "second label" are you suggesting we omit @konmouz ?

lurch, 2017-02-28:
Re-opening this issue...

PR #1193 (2017-05-10): refactor(GUI): central error/warning messages function
* Centralise drive error/warning messages with a function
  `.getDriveImageCompatibilityStatuses` in `lib/shared/drive-constraints.js` -- tests included
* Fix an error where several labels show at once
* Clarify the source drive label to 'Drive Contains Image'
* Remove label icons and make text bold to match Zeplin's design

Closes: #1143
Closes: #1144
Changelog-Entry: New central function as the source of drive error
messages.
closed 2017-05-10 by jviotti