balena-io/etcher #1153: Make images out of SD cards

Mo3tasm, 2017-03-02:
Great application, does what it says it does beautifully,

if it just can make Images from SD cards us backups or to image them to other SD cards (I know there's a suggestion to make Etcher copy an SD card to another on the flay, but sometimes you need to do it manually).

Thank a lot guys.

Mo3tasm, 2017-03-02:
Win32DiskImager does this the stupid way, by copying the whole SD card to hard disk, without worrying about how much data is actually being used and how much is free, but still, sometimes it's very useful.

jviotti, 2017-03-02:
Hi @Mo3tasm ,

Thanks for reaching out. We're tracking this feature at https://github.com/resin-io/etcher/issues/266. Keep an eye there for updates!

closed 2017-03-02 by jviotti