balena-io/etcher #1155: Publish etcher on chocolatey.org

pierrebeaucamp, 2017-03-03:
- **Etcher version:** 1.0.0-beta.19
- **Operating system and architecture:** Windows

Similar to Homebrew, [Chocolatey](https://chocolatey.org/) is the most used package manager for windows applications.

WasabiFan, 2017-03-03:
Related to https://github.com/resin-io/etcher/issues/829

jviotti, 2017-03-03:
Sounds interesting, and not hard to do, even though it'd be nice to get a hand from people used to creating Chocolatey packaged. Here's what's needed based on https://chocolatey.org/docs/create-packages. Please correct me if I'm missing something:

- Prepare the package. We distribute self-contained zips for Windows, so it looks like we can re-use those:

> If you have a self-contained package, you can remove the automation scripts
entirely and just include the runtime executables, they will automatically get shimmed,
which puts them on the path.

- Create a Nuspec configuration file
- Integrate the building routines into our build system (the Makefile + utility scripts). This includes:
  - Building the package with `cpack`
  - Uploading the package using Chocolatey

It'd be great if someone more familiar with Chocolatey can validate that the ZIPs that we create are good enough for publishing, and help us out with the Nuspec file :)

pierrebeaucamp, 2017-03-03:
I never created a package on chocolatey before, but since I opened this bug, I'd be happy to look into this some more this weekend and open a PR.

jviotti, 2017-03-03:
Sounds good, let us know how it goes!

pierrebeaucamp, 2017-03-06:
Turns out that building a package is pretty simple. Please take a look at https://github.com/pierrebeaucamp/chocolatey to see the necessary steps. I'd be happy to transfer the ownership of that repo to your account. 

I tested the scripts on a 64-bit Win10 machine, but I don't see why they would fail on a 32 bit machine.

jviotti, 2017-03-06:
That is amazing @pierrebeaucamp ! Thanks a lot for the awesome work. Don't bother with repository ownership transfer. I'll integrate the install script and the nuspec in the main repo as part of the build system at the beginning of the week :)

pierrebeaucamp, 2017-03-06:
Glad I could help. Two things to keep in mind:

* The nuspec I included is pretty basic. You can see a full list of supported fields here: https://docs.microsoft.com/en-ca/nuget/schema/nuspec. In particular, I didn't include the 'tags' field which lets people find the package more easily. 

* I don't know if your build system is running on Windows, but I think you will only be able to build & publish packages from a windows powershell.

jviotti, 2017-03-06:
Cool, I'll take a look at that and see what extra things I can include. I'll reference the PR here so you can take a final look. 

> I don't know if your build system is running on Windows, but I think you will only be able to build & publish packages from a windows powershell.

Yep, we build on Windows, so that shouldn't be a problem.

jviotti, 2017-03-06:
@pierrebeaucamp I've tested this and works like a charm. Just a thought: do you know if its possible to create a completely self contained `.nupkg` (e.g: that references local files instead of URLs)? We should do that if we can, so the `.nupkg` can be released independently, otherwise we have to make sure the installers have been uploaded to S3 before publishing to Chocolatey, or we will annoy a lot of people.

jviotti, 2017-03-06:
As a side note, PowerShell doesn't seem to be required. I was able to build just fine with `cmd.exe` as well.

pierrebeaucamp, 2017-03-06:
As far as I know, you can include any number of static files inside a nupkg. So yes, you could include the installer in the package itself.

However, I didn't find any clear instructions on how to bundle files with the package. Also, according to http://stackoverflow.com/a/32096709, it might be better not to bundle everything together as the installers are quite large.

jviotti, 2017-03-08:
Fair enough, let's make it reference external files. See https://github.com/resin-io/etcher/pull/1167

dragon788, 2017-04-15:
Just a note, if you are the original authors Chocolatey encourages packaging and bundling the installers (up to around 150MB total package size) as it makes it slightly faster for users to just download once instead of grabbing the nupkg and then having to wait for another download, but it definitely isn't required.

You could include the zip inside the nupkg, or you could just use the nuspec and in the `<files>` section you could have it recreate a specific folder structure if required for your app. Drop into https://gitter.im/chocolatey/chocolatey.org if you have any questions on packaging or best practices.

You can also use AppVeyor as your build system if you don't have access to a local CI pipeline.

jviotti, 2017-04-15:
Thanks a lot for the note @dragon788! I'm very interested in standalone packages, given that it makes our deployment story much simpler. We do use Appveyor for automated deployments, so we can definitely hook it there.

majkinetor, 2017-08-09:
Etcher is already available on chocolatey via my repository (since 1.0.0):

https://chocolatey.org/packages/etcher

It is automatically kept up to date and you can monitor updates via my [AU status page](https://gist.github.com/majkinetor/a700c70b8847b29ebb1c918d47ee4eb1)

I didn't embed the package because it was created during time when there was less then 150MB limit on package. Now its about 300 so I will embed it some time in the future.

jviotti, 2017-08-09:
Wow @majkinetor, this is amazing! Thanks for doing this! I'll update the instructions to point to your package :)

jviotti, 2017-08-09:
See https://github.com/resin-io/etcher/pull/1685

closed 2017-08-11 by jviotti