balena-io/etcher #1159: Error at launch when GRUB_CMDLINE_LINUX="net.ifnames=0"

actuino, 2017-03-06:
- **Etcher version:**
1.0.0. beta 19 x64
- **Operating system and architecture:**
Ubuntu Mate - 16.04.2 TLS
- **Do you see any meaningful error information on DevTools?**
Alert box on launch

When /etc/default/grub contains 
GRUB_CMDLINE_LINUX="net.ifnames=0"
Etcher shows an error at launch, before displaying:

```
Command failed: /tmp/.org.chromium.Chromium.KohxiJ
/tmp/.org.chromium.Chromium.KohxiJ: ligne 122: UDEV_net.ifnames=0 : commande introuvable

Error: Command failed: /tmp/.org.chromium.Chromium.KohxiJ
/tmp/.org.chromium.Chromium.KohxiJ: ligne 122: UDEV_net.ifnames=0 : commande introuvable

    at ChildProcess.exithandler (child_process.js:218:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```
Same error when launched in the devtool console log.
(The very same etcher was working on the same box before the Grub modification)
Image selection is then ok, but etcher doesn't recognize any usb drive.
Tried with the https://dl.bintray.com/resin-io/debian stable etcher repo too, idem.

lurch, 2017-03-06:
How strange. Sounds like it might be an Electron bug? Can you confirm that removing that GRUB setting makes Etcher work properly again?

Also, could you copy'n'paste the contents of `/tmp/.org.chromium.Chromium.KohxiJ` here please? (or whatever random filename you get when you see this problem next time)

jhermsmeier, 2017-03-06:
The `/tmp/.org.chromium.Chromium.KohxiJ` location looks like the place `drivelist` moves the shell scripts for drive detection to, and the error seems to indicate that the `udevadm` command's exported env vars don't come out the way they're supposed to: https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh#L81

lurch, 2017-03-06:
Yeah, I actually just found up https://github.com/resin-io-modules/drivelist/pull/125 which sounds very similar.

lurch, 2017-03-06:
@actuino Could you please paste here the result of
```bash
udevadm info --query=property --export --export-prefix=UDEV_ --name="/dev/sda"
```

Thanks.

actuino, 2017-03-06:
Here it is:  
```
UDEV_DEVLINKS='/dev/disk/by-id/wwn-0x5001b444a454f05e /dev/disk/by-id/ata-SanDisk_SDSSDA240G_161573402689 /dev/disk/by-path/pci-0000:00:1f.2-ata-1'
UDEV_DEVNAME='/dev/sda'
UDEV_DEVPATH='/devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sda'
UDEV_DEVTYPE='disk'
UDEV_ID_ATA='1'
UDEV_ID_ATA_DOWNLOAD_MICROCODE='1'
UDEV_ID_ATA_FEATURE_SET_APM='1'
UDEV_ID_ATA_FEATURE_SET_APM_ENABLED='0'
UDEV_ID_ATA_FEATURE_SET_HPA='1'
UDEV_ID_ATA_FEATURE_SET_HPA_ENABLED='1'
UDEV_ID_ATA_FEATURE_SET_PM='1'
UDEV_ID_ATA_FEATURE_SET_PM_ENABLED='1'
UDEV_ID_ATA_FEATURE_SET_SECURITY='1'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ENABLED='0'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ENHANCED_ERASE_UNIT_MIN='2'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ERASE_UNIT_MIN='2'
UDEV_ID_ATA_FEATURE_SET_SECURITY_FROZEN='1'
UDEV_ID_ATA_FEATURE_SET_SMART='1'
UDEV_ID_ATA_FEATURE_SET_SMART_ENABLED='1'
UDEV_ID_ATA_ROTATION_RATE_RPM='0'
UDEV_ID_ATA_SATA='1'
UDEV_ID_ATA_SATA_SIGNAL_RATE_GEN1='1'
UDEV_ID_ATA_SATA_SIGNAL_RATE_GEN2='1'
UDEV_ID_ATA_WRITE_CACHE='1'
UDEV_ID_ATA_WRITE_CACHE_ENABLED='1'
UDEV_ID_BUS='ata'
UDEV_ID_MODEL='SanDisk_SDSSDA240G'
UDEV_ID_MODEL_ENC='SanDisk\x20SDSSDA240G\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20'
UDEV_ID_PART_TABLE_TYPE='dos'
UDEV_ID_PART_TABLE_UUID='acfafa0d'
UDEV_ID_PATH='pci-0000:00:1f.2-ata-1'
UDEV_ID_PATH_TAG='pci-0000_00_1f_2-ata-1'
UDEV_ID_REVISION='Z22000RL'
UDEV_ID_SERIAL='SanDisk_SDSSDA240G_161573402689'
UDEV_ID_SERIAL_SHORT='161573402689'
UDEV_ID_TYPE='disk'
UDEV_ID_WWN='0x5001b444a454f05e'
UDEV_ID_WWN_WITH_EXTENSION='0x5001b444a454f05e'
UDEV_MAJOR='8'
UDEV_MINOR='0'
UDEV_SUBSYSTEM='block'
UDEV_TAGS=':systemd:'
UDEV_USEC_INITIALIZED='1801879'
UDEV_net.ifnames='0'
```

I'll then change Grub again and reboot to confirm.

jviotti, 2017-03-06:
Hm, so `awk -F= '{gsub("\\\\.","_",$1); print $1 "=" $2}'` is not taking effect?

@actuino Can you try:

```sh
udevadm info --query=property --export --export-prefix=UDEV_ --name="/dev/sda" | awk -F= '{gsub("\\\\.","_",$1); print $1 "=" $2}'
```

?

actuino, 2017-03-06:
Before rebooting, I checked the /tmp file, it's indeed the drive detection script
Here are the first lines:
```
#!/bin/bash

set -u
set -e

ignore_first_line() {
  tail -n +2
}

get_uuids() {
  /sbin/blkid -s UUID -o value "$1"*
}

get_mountpoints() {
  grep "^$1" /proc/mounts | cut -d ' ' -f 2
}

DISKS="$(lsblk -d --output NAME | ignore_first_line)"

for disk in $DISKS; do

  # Omit loop devices and CD/DVD drives
```

actuino, 2017-03-06:
@jviotti :
```
udevadm info --query=property --export --export-prefix=UDEV_ --name="/dev/sda" | awk -F= '{gsub("\\\\.","_",$1); print $1 "=" $2}'
UDEV_DEVLINKS='/dev/disk/by-id/wwn-0x5001b444a454f05e /dev/disk/by-id/ata-SanDisk_SDSSDA240G_161573402689 /dev/disk/by-path/pci-0000:00:1f.2-ata-1'
UDEV_DEVNAME='/dev/sda'
UDEV_DEVPATH='/devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sda'
UDEV_DEVTYPE='disk'
UDEV_ID_ATA='1'
UDEV_ID_ATA_DOWNLOAD_MICROCODE='1'
UDEV_ID_ATA_FEATURE_SET_APM='1'
UDEV_ID_ATA_FEATURE_SET_APM_ENABLED='0'
UDEV_ID_ATA_FEATURE_SET_HPA='1'
UDEV_ID_ATA_FEATURE_SET_HPA_ENABLED='1'
UDEV_ID_ATA_FEATURE_SET_PM='1'
UDEV_ID_ATA_FEATURE_SET_PM_ENABLED='1'
UDEV_ID_ATA_FEATURE_SET_SECURITY='1'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ENABLED='0'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ENHANCED_ERASE_UNIT_MIN='2'
UDEV_ID_ATA_FEATURE_SET_SECURITY_ERASE_UNIT_MIN='2'
UDEV_ID_ATA_FEATURE_SET_SECURITY_FROZEN='1'
UDEV_ID_ATA_FEATURE_SET_SMART='1'
UDEV_ID_ATA_FEATURE_SET_SMART_ENABLED='1'
UDEV_ID_ATA_ROTATION_RATE_RPM='0'
UDEV_ID_ATA_SATA='1'
UDEV_ID_ATA_SATA_SIGNAL_RATE_GEN1='1'
UDEV_ID_ATA_SATA_SIGNAL_RATE_GEN2='1'
UDEV_ID_ATA_WRITE_CACHE='1'
UDEV_ID_ATA_WRITE_CACHE_ENABLED='1'
UDEV_ID_BUS='ata'
UDEV_ID_MODEL='SanDisk_SDSSDA240G'
UDEV_ID_MODEL_ENC='SanDisk\x20SDSSDA240G\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20'
UDEV_ID_PART_TABLE_TYPE='dos'
UDEV_ID_PART_TABLE_UUID='acfafa0d'
UDEV_ID_PATH='pci-0000:00:1f.2-ata-1'
UDEV_ID_PATH_TAG='pci-0000_00_1f_2-ata-1'
UDEV_ID_REVISION='Z22000RL'
UDEV_ID_SERIAL='SanDisk_SDSSDA240G_161573402689'
UDEV_ID_SERIAL_SHORT='161573402689'
UDEV_ID_TYPE='disk'
UDEV_ID_WWN='0x5001b444a454f05e'
UDEV_ID_WWN_WITH_EXTENSION='0x5001b444a454f05e'
UDEV_MAJOR='8'
UDEV_MINOR='0'
UDEV_SUBSYSTEM='block'
UDEV_TAGS=':systemd:'
UDEV_USEC_INITIALIZED='1801879'
UDEV_net.ifnames='0'
```

lurch, 2017-03-06:
@jviotti I've just been doing some testing on the command line, and the `awk` appears to work correctly if I replace the 4-backslashes with just 2-backslashes. So maybe the fix really is that simple?

@actuino Could you download https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh and try running it, confirm that it fails with the same error, and then edit line 86 to remove two of the backslashes, and see if it works then? :-)

jviotti, 2017-03-06:
Oh, I see. If that's the case I wonder why I didn't catch it in my testing :/

actuino, 2017-03-06:
In the meantime, I rebooted without the net.ifnames=0 : etcher works well.
Re-rebooted with the line.

Tested the script, script fails with the same error.
Removed 2 of the backslashes, seems to work:
```
device: /dev/sda
description: "SanDisk SDSSDA24"
size: 240057409536
mountpoints:
  - path: /boot
raw: /dev/sda
protected: False
system: True
```

jviotti, 2017-03-06:
Awesome, thanks a lot for the testing @actuino, and sorry for the silly inconvenience :) @lurch Do you want to create the PR yourself?

actuino, 2017-03-06:
Hey, thank you for the fast troubleshooting, that's awesome !

jhermsmeier, 2017-03-06:
> Oh, I see. If that's the case I wonder why I didn't catch it in my testing :/

@jviotti Maybe different shells and/or shell, awk, gsub versions?

jviotti, 2017-03-06:
The script shebang is set to bash, so I guess its not because of different shells. @actuino thanks a lot :) As a last petition, do you mind sharing your awk version with us?

actuino, 2017-03-06:
Sure : `GNU Awk 4.1.3, API: 1.1 (GNU MPFR 3.1.4, GNU MP 6.1.0)`

jhermsmeier, 2017-03-06:
Oh, right – but possibly different bash versions? There were some significant changes in bash@4, if I remember it right.

actuino, 2017-03-06:
`GNU bash, version 4.3.46(1)-release (x86_64-pc-linux-gnu)`

lurch, 2017-03-06:
Yeah, if the fix is really so simple, I'm also wondering how this didn't get picked up in earlier testing.
I wouldn't expect something as simple as backslash-escaping to significantly change between bash versions? \*shrug\*

jviotti, 2017-03-06:
I'm running Ubuntu 16.04 as well, and:

- GNU bash, version 4.3.46(1)-release (x86_64-pc-linux-gnu)
- mawk 1.3.3 Nov 1996, Copyright (C) Michael D. Brennan

Maybe mawk vs GNU awk was the problem? @lurch if so, we should test that in mawk as well.

lurch, 2017-03-06:
Pinging @wrboyce as he's mentioned in https://github.com/resin-io-modules/drivelist/pull/125 - maybe he has some ideas :-)

lurch, 2017-03-06:
On Ubuntu 14.04, `awk`, `mawk` and `gawk` all behave identically with the awk-code used here.

lurch, 2017-03-07:
@jviotti Before we close this, could we do a snapshot-build for @actuino to confirm that this actually fixes his problem? (within Etcher)

jviotti, 2017-03-08:
Yeah, makes sense. I'll prepare one and upload it here.

jviotti, 2017-03-08:
Check https://drive.google.com/open?id=0B7tkbonGU-RyZFNzc01kS3ZCMWs

lurch, 2017-03-08:
Re-opening again ;-)

actuino, 2017-03-08:
Hi guys,
Sorry it took so long.
Got it and tested, no problem, works fine now : thanks a lot !

lurch, 2017-03-08:
Great news, thanks for confirming it's fixed @actuino :smiley:

closed 2017-03-08 by lurch