balena-io/etcher #1162: Bad Archive and Multiple Image Files Found Errors (NOOBS ZIP) (Linux)

MrDiamond123, 2017-03-08:
I am trying to install NOOBS on a SD Card to put it in my Raspberry Pi

Well the ZIP file I downloaded [http://director.downloads.raspberrypi.org/NOOBS/images/NOOBS-2017-03-03/NOOBS_v2_3_0.zip](url) was giving me a Bad Archive Error

and when I unzipped and re-zipped the file Etcher now gives me a Invalid archive image

Please Give Us A Option To Just Format and Unzip to USB/SD

MrDiamond123, 2017-03-08:
And Yes I know, I could do it myself but I (and some other people I guess) want a easy way of setting up their Pi and stuff...

jviotti, 2017-03-08:
Hi @MrDiamond123 ,

Thanks for reporting this. NOOBS is quite special since its not really an image, but a ZIP with various files, that should not be flashed to a drive (which is what Etcher is concerned with).

We could implement an option to format and unzip to USB, but as an image flasher, this is not something Etcher cares to put time into in the present nor in the near future.

That being said, I'd be more than happy to consider a PR from the community!

closed 2017-03-08 by jviotti

paul-hammant, 2017-10-11:
Etcher could easily have a mode of operation that's just for NOOBS installs:

1. Straight format the SD Card as FAT (not exFAT), and limit to max 32GB in case it is bigger.
2. Place files from the unzipped NOOBS download on the newly created FS. 
3. declare "done".

The SD card industry's standard formatter - https://www.sdcard.org/downloads/formatter_4/ - does exFAT by default and the Pi cannot boot that, and doesn't give the end user (often novices) any advice at all about what to do next.  Etcher purports to be an easy end-user tool to use. Special processing for NOOBS because Pi is such a major target wouldn't be too hard to add.

lurch, 2017-10-12:
Just to clarify the 32GB / exFAT issue... for cards smaller than 64GB, the official SD Formatter tool will format them as FAT32. For SD cards 64GB or larger, SD Formatter will format them as exFAT, because it considers them SDXC cards. There's more info about this here: https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

There's nothing stopping 64GB or 128GB SD cards being formatted as FAT32, but the formatting tools in Windows will refuse to format a disk as FAT32 if it's larger than 32GB. Therefore to use Windows to format a 64GB SDcard as FAT32 (so that it can be used with NOOBS) you have to use a 3rd-party formatting tool - this is also explained in the link above.

> Special processing for NOOBS because Pi is such a major target wouldn't be too hard to add.

We try our best to make Etcher as robust and fool-proof as possible. And as explained earlier, we can't rely on the native platform's (specifically Windows') formatting tools to format an SD card as FAT32, because the user might be using a 64GB SD card. Therefore we'd need to add our own custom FAT32-formatting code to Etcher, and this isn't a trivial task...

paul-hammant, 2017-10-12:
Maybe a warning dialog and a referral to a well written https://www.raspberrypi.org page on NOOBS and the whole FAT/exFAT/32GB non-image issue.  If if they don't have a well-written page, your own one perhaps?

bvrulez, 2017-10-12:
I tried to open NOOBS_v2_4_4.zip with Etcher and it gives an error. On [this](https://www.raspberrypi.org/documentation/installation/installing-images/) site installing Noobs with Etcher from a ZIP file is said to be possible. Not sure if it is a problem with Noobs or with Etcher. (I use a 128GB SD card which I formatted via command line as FAT32).

lurch, 2017-10-12:
@bvrulez That page you linked to is about installing disk images (which is exactly what Etcher is designed to do). NOOBS _isn't_ a disk image, and requires a different install procedure - which is why https://www.raspberrypi.org/documentation/installation/ has different instructions for "NOOBS" and "images".