balena-io/etcher #1165: Startup error

pfmscc, 2017-03-08:
- **Etcher version:** 1.0.0-beta.19
- **Operating system and architecture:** Win 10 Pro 64
- **Do you see any meaningful error information on DevTools?**

jviotti, 2017-03-08:
Can you provide more information about the issue?

pfmscc, 2017-03-08:
Can you see the image I uploaded?

It shows the error.

Wait, I’ll generate a new one.

[cid:image001.png@01D29831.CD1153D0]

If I click OK, it brings up the GUI.

I can select the ISO (though is appears to take to attempts sometimes) but I cannot select the drive.

Paul

EXTERNAL EMAIL

jviotti, 2017-03-08:
Hi @pfmscc ,

I don't think GitHub is showing your image at all. Maybe worth trying to upload it from the GitHub interface rather than from your email?

pfmscc, 2017-03-08:
My first attempt included it directly on github.

I don’t know what is blocking it.

I will type in the errors:

Command failed:
C:\Users\myusername\AppData\Local\Temp\9FEC.tmp.bat

  at ChildProcess.exithandler (child_process.js:218:12
  at emitTwo (events.js: 106: 13
  at ChildProcess.emit (events.js: 191: 7)
  at maybeClose (internal/child_process.js: 877: 16)
  atSocket.<anonymous> (internal/child_process.js:334: 11)
  at emitOne (events.js:96: 13)
  at Socket.emit (events.js: 188: 7)
  at Pipe._handle.close [as _onclose] (net.js:493: 12)

I have reinstalled twice with no change in behavior.

The .bat file name changes with each iteration.

Paul

EXTERNAL EMAIL

jviotti, 2017-03-08:
Awesome, thanks for the detailed error. This looks to be the same problem as  https://github.com/resin-io/etcher/issues/1054, so I'll close this one and we can continue discussion there.

Do you mind running the following `.bat` script on your computer, and see how that goes? https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat

That is the same bat script that gets you the error. I'd also be great if you can do `echo %ERRORLEVEL%` right after executing the script and report back the integer that you get.

closed 2017-03-08 by jviotti

pfmscc, 2017-03-09:
I ran the script and put the results in #1054<https://github.com/resin-io/etcher/issues/1054>.

Paul

EXTERNAL EMAIL