balena-io/etcher #1166: "Connect a drive" has no options

Ruffturn, 2017-03-08:
- 1.0.0-beta.19
- Win7
- Don't know what this is:**Do you see any meaningful error information on DevTools?**
Doesn't help:

jviotti, 2017-03-08:
Hi @Ruffturn,

"Connect a drive" is not a button. Etcher tries to filter out non removable drives by default, so it will avoid showing your C drive and your external USB drive, however the thumb drive should be detected just fine. If for some reason Windows is detecting it as a non removable drive, you can enable "Unsafe mode" in the setting page (the cog icon in the top right corner), which will disable such check.

Ruffturn, 2017-03-09:
I'll try that. Thanks.

Sent via the Samsung Galaxy S7, an AT&T 4G LTE smartphone

jviotti, 2017-03-09:
In case that works, do you mind running the following script on your computer while the problematic drive is plugged in and report back the results? https://github.com/resin-io-modules/drivelist/blob/master/scripts/win32.bat

jviotti, 2017-03-15:
@Ruffturn Did you get a chance to run the script?

jviotti, 2017-03-24:
I'll close this issue, given we don't have any information to proceed. Please re-open if you find the time to run the script linked above.

closed 2017-03-24 by jviotti