balena-io/etcher #1170: Show friendly name for removable drive in main window

dlech, 2017-03-09:
- **Etcher version:**
beta19
- **Operating system and architecture:**
all of the above

It makes me really nervous flashing a drive when it just says `/dev/sdc`. I always double check by clicking "change", which lists slightly more drive information or I use another application to confirm that the drive is really what I think it is. It would be nice to do this with fewer clicks. I would rather see the "friendly" name instead of the path in the main window (and maybe show the path on hover).

lurch, 2017-03-09:
Pinging @konmouz , our UX designer.

I guess the trouble with showing _only_ the "friendly name" is that you might have multiple drives that appear with the same "friendly name", whereas the raw-drive-name is guaranteed to be unique. So IMHO we should always at least show the raw-drive-name, and maybe also show the friendly-name at the same time?

konmouz, 2017-03-09:
[image]
[image]

jviotti, 2017-03-09:
@konmouz Its easy to show the friendly name there, however notice that we need to deal with how long names get displayed/truncated. Any ideas?

konmouz, 2017-03-09:
We should used ellipsis "..." here as well. We can put the size (e.g. 15.8GB) in a new line, but this will be a bit confusing with the CTA just below (might look like another CTA to some users).

dlech, 2017-03-09:
What if you put the ellipsis in the middle, like `really lo...ng name`.

lurch, 2017-03-09:
> What if you put the ellipsis in the middle, like `really lo...ng name`?

That's exactly how the ellipses (in Etcher) already work, see the `resin-tes...afe02.img` in the above screenshot ;-)

jviotti, 2017-03-09:
As a fun fact, the ellipses module we're using was tested with JSVerify (http://jsverify.github.io) with great success. We should bring that to more Etcher components :)

Shou, 2017-03-15:
So I've got this PR now. I don't think it's the ideal solution though—I agree with @lurch's comment above in that we should at least show the mount point (considering it's truly unique). I can't think of a nice way to resolve it design-wise—both would take up too much space together as it is right now, hiding the mount point in a popover can still cause users to mistakenly flash one of two equally named devices, and hiding the device name in a popover instead almost brings us back to where we are now.

What do you guys think?

dlech, 2017-03-15:
Given the fact that the mount point is less meaningful to most people, I would put it in the popover. I suspect that having to identical devices plugged in at the same time is rare.

jviotti, 2017-03-15:
What do you think @konmouz ?

konmouz, 2017-03-16:
Having 3 pieces of info exposed is too much (name, mount point, size). For me mount point can be on the tooltip or more info, but if you think that this is high priority we can hide the drive size.

konmouz, 2017-03-16:
It is good to solve this issue now by prioritizing the info so we can use the same solution on v02.

jviotti, 2017-03-16:
The size is probably more important. I think the name + size is the best combination.

lurch, 2017-03-21:
Can we keep this open until the tooltip part gets added?

jviotti, 2017-03-24:
I'm closing this, since it turns out we already had a tooltip there :)

lurch, 2017-03-27:
Re-opening, because in #1181 @konmouz suggested also showing the non-truncated drive description (which might get displayed with ellipses if it's too long) in the tooltip ;)

closed 2017-03-29 by jviotti