balena-io/etcher #1179: "Error: No polkit authentication agent found" when trying to flash an image as non-root

soulshake, 2017-03-11:
- **Etcher version:** 1.0.0-beta.19
- **Operating system and architecture:** 
```
$ uname -a
Linux patamushka 4.9.0-997-generic #201612270045 SMP Tue Dec 27 05:47:01 UTC 2016 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (stretch)
Release:	9.0
Codename:	stretch
```

## What I do

- open Etcher via CLI by running `etcher-electron` (as my normal system user)
- select an image I've just downloaded from resin.io
- (the disk autopopulates with my SD card)
- click "Flash"

## What I expect to happen

For the image to be written to the card

## What actually happens

I get the following popup:

[image]

```
{ Error: No polkit authentication agent found.
    at /usr/share/etcher-electron/resources/app.asar/node_modules/sudo-prompt/index.js:132:23
    at ChildProcess.exithandler (child_process.js:225:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
  cause: 
   Error: No polkit authentication agent found.
       at /usr/share/etcher-electron/resources/app.asar/node_modules/sudo-prompt/index.js:132:23
       at ChildProcess.exithandler (child_process.js:225:5)
       at emitTwo (events.js:106:13)
       at ChildProcess.emit (events.js:191:7)
       at maybeClose (internal/child_process.js:877:16)
       at Socket.<anonymous> (internal/child_process.js:334:11)
       at emitOne (events.js:96:13)
       at Socket.emit (events.js:188:7)
       at Pipe._handle.close [as _onclose] (net.js:493:12),
  isOperational: true }
```

```
    at Socket.child.stderr.on (/usr/share/etcher-electron/resources/app.asar/lib/child-writer/index.js:129:17)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at Pipe.onread (net.js:543:20)
```

After I click OK, another popup appears:

```
Attention
Oops, seems something went wrong.
[Retry]
```
Retrying produces the same result.

At the same time as the "Attention" popup, this appears in my system logs:

```
Mar 11 10:42:30 patamushka dunst[7910]: g_variant_unref: assertion 'value->ref_count > 0' failed
Mar 11 10:42:30 patamushka dunst[7910]: g_variant_unref: assertion 'value->ref_count > 0' failed
Mar 11 10:42:30 patamushka dunst[7910]: g_variant_unref: assertion 'value->ref_count > 0' failed
```

And this in my `auth.log`:

```
Mar 11 10:42:24 patamushka sudo:       aj : a password is required ; TTY=pts/1 ; PWD=/home/aj ; USER=root ; COMMAND=/usr/bin/env ELECTRON_RUN_AS_NODE=1 IPC_SERVER_ID=etcher-server-9763 IPC_CLIENT_ID=etcher-client-9763 SKIP=1 /usr/share/etcher-electron/etcher /usr/share/etcher-electron/resources/app.asar/lib/child-writer/writer-proxy.js /usr/share/etcher-electron/resources/app.asar /home/aj/Downloads/resin-blinky-1.24.1-2.8.3-eef8cf4afe02.img --drive /dev/mmcblk0 --unmount --check
```

## Workaround

By sheer chance, I noticed that if I run a command with `sudo`--any command, e.g. `sudo ls`--I'm then able to flash successfully, even though I'm not running `etcher-electron` with sudo.

If I run via `sudo etcher-electron` everything works.

jviotti, 2017-03-11:
Hi @soulshake,

Etcher requires a Polkit authentication agent to be present on the OS (which comes by default in virtually all major distributions). 

The debian package requires a polkit authentication agent in order to be installed (https://github.com/resin-io/etcher/blob/master/scripts/build/debian/config.json#L43). Could it be that you installed the Etcher deb, but then removed all available polkit agents?

You should be fine after installing `polkit-1-auth-agent`,  `policykit-1-gnome, or `polkit-kde-1`. You can probably google other authentication agents for less known desktop environments.

Etcher worked fine with `sudo` given that if you have enough permissions already, then the elevation system is not ran.

jviotti, 2017-03-11:
Actually, let's keep this open and send a PR that improves such unfriendly error message.

jviotti, 2017-03-11:
Maybe @lurch or @Shou can send a quick PR for this?

soulshake, 2017-03-12:
Thanks for the explanation, makes sense now. I'm using i3.
FYI, I already had `policykit-1-gnome` installed (not the two others though).
I installed [`mate-polkit-bin`](https://packages.debian.org/stretch/mate-polkit-bin) and it works perfectly now after a reboot.

RogueM, 2017-03-16:
I'm running Debian sid/stretch in a chroot. I have lxsession/lxpolkit installed, running and working. I installed lxqt-policykit manually and launched it to try to work around the message described herein. No dice.

My workaround was the same as @soulshake, `sudo ls` to temporarily allow operations with elevated privileges. I was then able to flash an image then using the Etcher GUI.

lurch, 2017-03-16:
Hello @RogueM , who let you in here?? ;-)
Maybe there's a problem that https://github.com/jorangreef/sudo-prompt (or perhaps some other part of the stack) isn't aware of all the polkit agents that it should be?

RogueM, 2017-03-16:
no, I think I have narrowed down the issue to the chroot, at least in my case... some agents are motre explicit about the issue and I don't think I can make any active. In fact, now that I now about the issue I have tried to run `synaptic` and it forever leaps into an oblivion. running `synaptic-pkexec` from cli all works fine, after password prompt.

so conclusion: I don't think it's a problem either projects can/should address.

jviotti, 2017-03-24:
@Shou I remember you were working on this at some point. Any updates? Did you get stuck somewhere?

Shou, 2017-03-24:
I did get somewhere but I didn't quite finish it, I got preoccupied with other things. I'll probably return to it after the weekend—the information you gave me in the chat was very adequate so I can't remember having any questions, but I'll get back to you if I do.

jviotti, 2017-03-24:
@Shou Is there a branch somewhere with the work in progress? I rather have you focusing on v2, so I can quickly finish it if you like.

Shou, 2017-03-24:
Looking at it there wasn't a significant amount of work done here because I was trying to figure out how to use robot, so you can go ahead, I don't think you'll miss the work I did.

jviotti, 2017-03-24:
Awesome, I just sent a PR for it!

breznak, 2018-09-01:
I'd like to reopen this bug for Ubuntu 18.04.1, with KDE Neon. 
I get the err message (nicer format), even though polkit is installed: 
`polkit-kde-agent-1 is already the newest version (4:5.13.4-0xneon+18.04+bionic+build11)`

Running etcher from your official repo, stable channel

jhermsmeier, 2018-09-03:
@breznak thanks, we'll check it out, and reopen & fix if we find something! Meanwhile, can you post the full error message you're getting?

SBanya, 2019-10-09:
I'm getting the same error when trying to write Debian AMD-64 to a standard 64 gig flash drive.

The only work around is to elevate to sudo user, and then run Etcher.

tkedwards, 2020-10-30:
I'm getting the same issue on Ubuntu 18.04. Polkit is installed:
```
ii  libpolkit-agent-1-0:amd64                     0.105-20ubuntu0.18.04.5                             amd64        PolicyKit Authentication Agent API
ii  libpolkit-backend-1-0:amd64                   0.105-20ubuntu0.18.04.5                             amd64        PolicyKit backend API
ii  libpolkit-gobject-1-0:amd64                   0.105-20ubuntu0.18.04.5                             amd64        PolicyKit Authorization API
ii  libpolkit-qt-1-1:amd64                        0.112.0-5                                           amd64        PolicyKit-qt-1 library
ii  libpolkit-qt5-1-1:amd64                       0.112.0-5                                           amd64        PolicyKit-qt5-1 library
ii  polkit-kde-agent-1                            4:5.12.7-0ubuntu0.1build1~ubuntu18.04~ppa1          amd64        KDE dialogs for PolicyKit
```

I'd guess that because this program runs from an AppImage it's somehow not able to find or use the 'system' polkit? Anyway, it's definitely still a bug

Running with sudo works around it, e.g:
sudo ./balenaEtcher-1.5.109-x64.AppImage

acestronautical, 2021-11-16:
chrome remote desktop caused this for me :<

sidml, 2021-11-27:
I got this error on i3. Running AppImage using `sudo` seems to work.

Vlek, 2022-04-24:
I just got this on Arch/i3. As @sidml mentioned, elevating privileges worked.

lurch, 2022-04-26:
This is a very old issue, but see https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#no-polkit-authentication-agent-found-error-in-gnulinux for more info.

kagutaba256, 2022-08-06:
Wish the error message was a bit more clear, `sudo` worked for me after finding this thread

mrquincle, 2023-01-10:
Just check what kind of agent is on your system and actually run it. For example I'm using `i3`, but I still have a polkit agent on my system. It just hasn't started.

Run it in a different shell:

```
/usr/lib/policykit-1-gnome/polkit-gnome-authentication-agent-1
```

Place it in `~/.xinitrc` if you like to have it always run, otherwise just remember to actually run the agent. :-)

No need to use `sudo`.

ethagnawl, 2023-05-23:
This is a reminder to my future self and others that if you do need to run Etcher using sudo, you may need to provide an absolute path to the executable. 

```
# does not work
$ sudo balenaEtcher-1.7.1-x64.AppImage 
sudo: balenaEtcher-1.7.1-x64.AppImage: command not found

# works

```

Vlek, 2023-05-26:
I want to also leave a note for myself that Balena Etcher is on the list of not recommended programs in the Arch Wiki and to instead use other options.

conarwelsh, 2023-06-18:
for anyone finding this...
this happened to me after installing google remote desktop, and the fix was just permissions so this was the fix for me:

```
sudo chmod 5755 /usr/lib/policykit-1/polkit-agent-helper-1
```

mangelozzi, 2023-07-04:
Ubuntu 22, tried `mate-polkit-bin`, `polkit-1-auth-agent`, `policykit-1-gnome` ... none of them fixed the issue

ethagnawl, 2023-07-06:
I've found myself back at this thread after running into this issue again (Debian 12 and i3) and I can confirm that @mrquincle's suggestion to manually start polkit results in my being able to run the AppImage without requiring sudo.

UTFeight, 2023-07-11:
I use Hyprland as my window manager (requires polkit-kde) and you have to start polkit automatically with this config command:
```py
exec-once = /usr/lib/polkit-kde-authentication-agent-1
```
or you can simply run it for once from commandline like so 
```shell
exec /usr/lib/polkit-kde-authentication-agent-1
```

fardeenpanjwani-codeglo, 2023-10-22:
> Just check what kind of agent is on your system and actually run it. For example I'm using `i3`, but I still have a polkit agent on my system. It just hasn't started.
>
> Run it in a different shell: [...]

This finally worked for me on ub-22.04/i3

UTFeight, 2023-11-11:
```
Listener adapter polkit_qt_listener_initiate_authentication
GSimpleAsyncResult: 0x55d09364b450
polkit_qt_listener_initiate_authentication callback for  0x55d093635f10
Initiating authentication
Cannot mix incompatible Qt library (5.15.11) with this library (5.15.10)
KCrash: Attempting to start /usr/lib/polkit-kde-authentication-agent-1
KCrash: crashing... crashRecursionCounter = 2
KCrash: Application Name = polkit-kde-authentication-agent-1 path = /usr/lib pid = 97058
KCrash: Arguments: /usr/lib/polkit-kde-authentication-agent-1
fish: Job 1, '/usr/lib/polkit-kde-authenticat…' terminated by signal SIGABRT (Abort)
```

I get a segfault ?