balena-io/etcher #1180: Processes duplicated: conhost, xxx.tmp.exe, etcher.exe

Dec4pt0x1, 2017-03-14:
**Etcher version:** v1.0.0-beta.19
**Operating system and architecture:** Windows 10 x64

This happened when imaging Raspbian to MicroSDcard (Sandisk MobileMate Card Reader)
Had to reboot, unable to ternimate the constant process spam executions. Viewed in Process Explorer.
[image]

lurch, 2017-03-14:
Looks like some process within Etcher is failing (maybe the elevation?), and it's attempting to restart that process to get around that problem (which is failing again?) leading to this deeply-nested process tree.
Do you see any other erroneous behaviour in Etcher? Apart from these multiple-processes, did Etcher actually work correctly?
Do you see anything logged in DevTools? (press `Ctrl+Alt+I` while in the Etcher GUI)

jviotti, 2017-03-14:
Hi @Dec4pt0x1,

Can you try running `net.exe session` on `cmd.exe` and report back what you get? I'm thinking you're hitting this issue: https://github.com/sindresorhus/is-admin/issues/3

Dec4pt0x1, 2017-03-15:
@jviotti Yes, that was disabled a couple days ago, human error on my end :) Problem solved.

jviotti, 2017-03-15:
I'm glad we could confirm that was the issue. I sent a PR to the referenced module to fix this issue (https://github.com/sindresorhus/is-admin/pull/4), but in the mid-time you can either enable such service, or run Etcher as administrator. Sorry for the inconveniences!

lurch, 2017-03-15:
@jviotti https://github.com/sindresorhus/is-admin/pull/4 has now been merged, and https://www.npmjs.com/package/is-admin has been bumped to 2.1.0 :tada: :tada: 

I won't submit a PR upgrading `package.json` myself, as my Windows laptop isn't yet set up to build Etcher, so I wouldn't be able to verify that this issue is actually fixed by the newer module.

jviotti, 2017-03-15:
Great news! We'll first need to bump the version at `is-elevated` (https://github.com/sindresorhus/is-elevated). I'll send a PR for it.

jviotti, 2017-03-15:
Here's the PR: https://github.com/sindresorhus/is-elevated/pull/1

lurch, 2017-03-15:
`is-elevated` 2.0.0 is now available :-)

jviotti, 2017-03-15:
`is-elevated` is broken, given that it doesn't return a promise for OS X and GNU/Linux. Let's wait for the next release.

closed 2017-03-21 by jviotti