balena-io/etcher #1195: "Connect a drive" portion is not working?

poohomar, 2017-03-21:
- **Etcher version:**
- **Operating system and architecture:**
- **Do you see any meaningful error information on DevTools?**

poohomar, 2017-03-21:
"Connect a Drive" portion is not working after you select image.

jviotti, 2017-03-21:
Hi @poohomar ,

"Connect a drive" is not a button. Etcher might be failing to recognise your drive, or it might think its not a removable drive, and therefore protecting you from writing to it.

Can you enable "unsafe mode" in the settings page (the cog in the top right corner) and see if that makes a difference?

If so, what operating system are you running?

poohomar, 2017-03-21:
Hello Jviotti,

I followed your instructions and it somewhat worked, now I can access the driver portion. But it won't allow me to select my DVD/CD bay.

[image]

lurch, 2017-03-21:
Etcher doesn't write to optical disks (DVDs or CDs), those require a specific CD/DVD-burning program (I thought we had an issue discussing that but I can't find it now).

Etcher writes images to USB flash drives, SD cards, etc.

closed 2017-03-21 by lurch