balena-io/etcher #1198: Ability to submit auto-filled issues from Etcher

Shou, 2017-03-22:
Something like a button or other that when triggered opens a link to a new Github issue, partially filled with information from their local Etcher. Such information can be the version string, system, processor architecture, and more.

Example: https://github.com/resin-io/etcher/issues/new?title=foo&body=bar

lurch, 2018-02-26:
Perhaps this could tie in with the work done in #1888 ?

jhermsmeier, 2018-02-26:
Yeah, indeed it would be nice to have a bug-reporter in Etcher that will also bundle some more data and could open an issue with all of it attached. Could also add a report generated by [node-report](https://github.com/nodejs/node-report) or so

closed 2018-10-15 by jviotti