balena-io/etcher #1200: Makefile: Operating system and architecture detection

kyrios123, 2017-03-22:
- **Etcher version:** 1.0.0-beta.19
- **Operating system and architecture:** Solus (Linux x86_64)
- **Do you see any meaningful error information on DevTools?** N/A

When I try to compile from the source, I get the following error : 
`Makefile:86: *** We couldn't detect your host architecture.  Stop.`

I had a look at the make file and I see architecture detection is done with _uname **-p**_
```
ifeq ($(shell uname -p),x86_64)
	HOST_ARCH = x64
endif
```

I think it would be safer to use _uname **-m**_

```
$ uname -p
unknown
$ uname -m
x86_64
```

jviotti, 2017-03-22:
Hi @kyrios123 ,

Thanks a lot for the heads up. Would you like to send a PR yourself? :)

kyrios123, 2017-03-22:
I could but I have only tested on my own computer so perhaps it would be safer to first check if `uname -m` gives the same result on other major linux distributions

jviotti, 2017-03-22:
I suggest to try `uname -p` first, given that is what everyone is using, and fallback to `uname -m` if `uname -p` returns `unknown`. What do you think? 

Maybe @lurch @Shou can check this as well?

kyrios123, 2017-03-22:
@jviotti 
Perhaps `uname -m` would work on all linux distributions (would be the best case scenario). If not I guess your proposal would be the way to go.

I just checked on a VM running Linux Mint. There both uname -p and -m return x86_64

Just in case, I'll raise an issue on Solus side to know if getting unknown is the expected result to uname -p

lurch, 2017-03-22:
`man uname` says:
```
       -m, --machine
              print the machine hardware name

       -p, --processor
              print the processor type or "unknown"

       -i, --hardware-platform
              print the hardware platform or "unknown"
```
so I agree with @jviotti 's suggestion (on my Ubuntu 14.04 they _all_ report `x86_64`). Although as `-m` above _doesn't_ say `or "unknown"`, then perhaps it _is_ the more reliable option? 

....

Hmmm, I've just read through all of https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=193170 and I'm now inclined to agree that it should be `uname -m` as it's [POSIX compliant](http://pubs.opengroup.org/onlinepubs/9699919799/utilities/uname.html).

Feel free to submit a PR @kyrios123 :+1:

kyrios123, 2017-03-23:
Here is the [pull request](https://github.com/resin-io/etcher/pull/1202).

lurch, 2017-03-23:
Closed by #1207

closed 2017-03-23 by lurch