balena-io/etcher #1210: Disallow reloading the renderer process

jviotti, 2017-03-23:
Users can reload the renderer process with Cmd+R and Ctrl+R. We should disallow this, given that it would result in undefined behavior, and weird errors.

Shou, 2017-03-23:
Is this platform-specific? I can't reload Etcher with Ctrl+R on Ubuntu, only from within the developer tools.

jviotti, 2017-03-24:
Hm, maybe F5 in that platform? I'll check here.

Shou, 2017-03-24:
On Ubuntu F5 only works from within the developer tools too.

jviotti, 2017-03-24:
You're right. It only happens on OS X. Before changing the issue tag to `platform:os x`, do you think there is a way to disable reloading from DevTools as well?

I think we should force reloading by re-running the app, given that refreshing, even from DevTools, can cause strange error that will get into TrackJS.

jviotti, 2017-03-24:
I'll fix it for the main Etcher window. We can't do much with DevTools since we have little control over it.

alexandrosm, 2017-03-26:
How about making the UI reload-tolerant? Is it a big job?

jviotti, 2017-03-26:
Each renderer process has a unique IPC identifier so that the child writer process can communicate with it. Making the UI reload-tolerant would involve storing that runtime calculated unique identifier somewhere where it could be matched with the reloaded renderer process but still not collide with other running instances of Etcher. On top of that, we would have to somehow restore the application state based on the fact that we detect that the child writer is running.

So in summary, it could be done, but its a lot of work. I don't think its worth the effort.

closed 2017-03-26 by jviotti