balena-io/etcher #1225: Flickering On The Screen

Eliteguardians, 2017-03-25:
- **Etcher version:**
- **Operating system and architecture:**
- **Do you see any meaningful error information on DevTools?**

     Hi I was wondering how do i remove this key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61 and the update from the linux mint 17.3. rosa?

jviotti, 2017-03-26:
Hi @Eliteguardians ,

I'm afraid I don't understand the issue. The title describes a flickering on the screen, but the description mentions some PGP issue and a Linux Mint 17.3 update. Can you clarify?

lurch, 2017-03-28:
I wonder if @Eliteguardians is asking how to "undo" all the steps in https://github.com/resin-io/etcher/blob/master/README.md#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 ?

It should simply be a case of:
```bash
sudo apt-get remove etcher-electron
sudo apt-key del D401AB61
sudo rm /etc/apt/sources.list.d/etcher.list
sudo apt-get update
```

Although AFAIK Etcher shouldn't make your screen flicker, so that might be being caused by some other update you installed.

jviotti, 2017-03-28:
I see. I sent a PR to add uninstall instructions to the README.

closed 2017-03-29 by jviotti

lurch, 2017-03-29:
@Eliteguardians Feel free to comment and re-open this issue if it doesn't solve your problem.