balena-io/etcher #1232: Drag and drop doesn't use the manifest title

Shou, 2017-03-29:
When dragging and dropping a ZIP with a manifest name, the filename is used instead. Using the normal file selector dialog doesn't cause this to happen.

Screenshots of drag and drop versus through the file selector.
[image]
[image]

jviotti, 2017-03-29:
Ugh, looks like we should make sure we use the same logic in both scenarios.

jviotti, 2017-03-29:
I think I coincidentally solved this issue while fixing drag and drop for https://github.com/resin-io/etcher/pull/1229. Do you mind pulling my latest changes from that PR and confirming that it solves the issue for you too? If so, I'll add a "Fixes" footer tag to the commit.

lurch, 2017-03-29:
Wow, I never imagined my [refactoring suggestion](https://github.com/resin-io/etcher/pull/1223#discussion_r108165110) would lead to fixes in the drag'n'drop code too. Awesome! :-D

jviotti, 2017-03-29:
Don't be humble, you knew that from the start! :P

lurch, 2017-04-04:
@Shou Is this indeed resolved by #1229 ? If it is, should this PR #1232 be closed?

Shou, 2017-04-05:
Yup. We can close this once it gets merged.

jviotti, 2017-04-05:
I added a "Fixes" tag to the PR.

closed 2017-04-06 by jviotti