balena-io/etcher #1249: Improve error message when a ZIP archive contains no images

jviotti, 2017-03-30:
Its usually the case that the user is selecting a plain ZIP archive which contains things other than images. Maybe we can improve the error message by explaining that ZIP archives are expected to contain image files, and not any files inside?

lurch, 2017-03-30:
What about something like "Selected zip archive contains multiple files. Etcher only works with zip archives containing a single disk image file." ?  Still not great, but possibly better than the current "The archive image should contain one and only one top image file".

Ping @alexandrosm

mrjcleaver, 2017-05-20:
Agreed, the message wasn't helpful when trying to use Etcher to install https://github.com/raspberrypi/noobs !

christiantreutler-avi, 2017-07-18:
Ran into this problem today. Downloaded the latest Raspbian and tried to burn the ZIP file. The exception is not human readable and I could not understand why it failed. This needs much better error handling!

lurch, 2017-07-18:
@christiantreutler-avi I've just downloaded and flashed the latest Raspbian image `2017-07-05-raspbian-jessie.zip` myself with Etcher, and it went absolutely fine. So I suspect you may be suffering from some other problem? Please open a new issue and provide as many details as you can, and we'll help you investigate.

lurch, 2017-07-18:
But regarding the original topic of this issue (the wording when selecting a zip archive containing more than one file), pinging @alexandrosm and @konmouz for their thoughts.

christiantreutler-avi, 2017-07-18:
@lurch I used the lite image `2017-07-05-raspbian-jessie-lite.zip` When I unpacked the ZIP file it worked. Can't really try it again, because my SD card is now full and running ;)

lurch, 2017-07-18:
I've just tried flashing `2017-07-05-raspbian-jessie-lite.zip` and (as expected) that went fine too! So I've got no idea what problem you might have been having. But if it ever happens again, feel free to open a new issue and we'll be happy to help :)

jviotti, 2017-07-18:
@christiantreutler-avi Can you give us a screenshot of what you get?

lurch, 2017-07-18:
@jviotti In his last message he said
> Can't really try it again, because my SD card is now full and running ;)

christiantreutler-avi, 2017-07-18:
@lurch Tried it again. Cannot reproduce with another SD card. Whatever it was, not sure. I like Etcher, so I will use it more. Maybe I get lucky - then I do a screenshot.

closed 2018-10-15 by jviotti