balena-io/etcher #1292: More room for showing filename and disk name?

dlech, 2017-04-12:
- **Etcher version:** 1.0.0-rc2
- **Operating system and architecture:** Ubuntu 16.04 x64 
- **Do you see any meaningful error information on DevTools?** N.A

We could make more room for displaying the image file name and the disk name if we moved the sizes below instead of on the same line.

In the example below, my drive name is "STORAGE DEVICE", which is not unreasonably long, so it would be nicer if it was not truncated.

[image]

jviotti, 2017-04-12:
Thanks for the idea @dlech. Let's bring @konmouz on the loop. Maybe we can improve this somehow on the v2 design?

konmouz, 2017-04-13:
@jviotti, Indeed on v2 we don't have this issue. The old solution for that was to use a two line system. The only concern here is that the 'change' button wouldn't appear as clear as it appears now (there will be 3 lines: name - size - change CTA). I'll upload a screenshot to see how it might look.

konmouz, 2017-04-13:
[image]

jviotti, 2017-04-13:
Sounds good @konmouz. I'll assign @Shou in case he has some time to tweak the v1.0 design as proposed, but definitely not very high priority if this is getting fixed on v2 already.

closed 2018-10-15 by jviotti