balena-io/etcher #1293: Retry a couple of times before giving up on ENXIO

jviotti, 2017-04-12:


lurch, 2017-04-13:
I saw you mention on gitter that this might be caused by a loose SD card. If that's the case, I guess it's possible that the data we previously tried writing to the card has now been corrupted, in which case is a retry the best course of action?

jviotti, 2017-04-13:
Hm, good point. We do know which block caused the ENXIO, so we can retry that one and continue, assuming everything before that was read/written correctly. If that's not the case, the validation process will eventually fail, so I guess its fine. What do you think?

lurch, 2017-04-20:
Yeah, I guess as long as the validation will catch it, we should be good.
Hmmm, if the user has validation turned off, should we maybe then *not* perform auto-retries, just to be on the safe side? :-/

jviotti, 2017-04-20:
That is a rough probability of the data being corrupted after ENXIO that you can think of? If this happens less than 50% of the cases, maybe we should still retry? After all, if the user didn't enable validation, then its because the user understands that these things could happen.

jviotti, 2017-04-20:
What do you think @jhermsmeier ?

lurch, 2017-04-20:
I've never seen ENXIO myself (I guess that's the problem with only buying high-quality equipment :wink: ), so I've got absolutely no idea what the chances of corruption are :-S
(Assuming you meant "What is..." instead of "That is..."?)

I suspect that for the users that Etcher is targeted at, most of them will probably just turn off validation because "it takes too long", rather than because they're prepared to accept getting errors on their drives ;)  Do we have statistics about what proportion of users turn off validation?

jviotti, 2017-04-20:
> I guess that's the problem with only buying high-quality equipment 

lol, its not too late to fix that "problematic" behavior :P 

> Do we have statistics about what proportion of users turn off validation?

The flash event includes a boolean property to determine that, which we could use to check (https://github.com/resin-io/etcher/blob/master/lib/gui/pages/main/controllers/flash.js#L66). Do you have access to Mixpanel?

lurch, 2017-04-20:
> The flash event includes a boolean property to determine that, which we could use to check (https://github.com/resin-io/etcher/blob/master/lib/gui/pages/main/controllers/flash.js#L66).

Huh, what does `unmountOnSuccess` have to do with whether the user has turned off validation or not?

> Do you have access to Mixpanel?

Nope, not yet.

jviotti, 2017-04-20:
Damn, I thought I said validate on success. We should add that then... I'll give you access to Mixpanel in a second...

jviotti, 2017-04-20:
Check https://github.com/resin-io/etcher/issues/1333

lurch, 2018-02-26:
@jhermsmeier Did #1885 fix this?

jhermsmeier, 2018-02-26:
Yes, indeed it did!

closed 2018-02-26 by jhermsmeier