balena-io/etcher #1305: Ubuntu 17.04

richprim, 2017-04-14:
- **Etcher version:** 1.0.0.  build 19
- **Operating system and architecture:** Ubuntu 17.04 on a 64 bit HP computer
- **Do you see any meaningful error information on DevTools?**
Etcher won't work on the new Ubuntu 17.04 OS
Rich Prim

jviotti, 2017-04-15:
Hey there @richprim,

Thanks for reporting! I'll install Ubuntu 17.04 next week and I'll see how it goes. What is the error that you get though?

richprim, 2017-04-18:
Etcher just doesn't open.
It asked if it should open it on the desktop, I click yes and noting happens.
Etcher is a great program, much easier then dd.
I had put 17.04 on a second drive so just re-installed my 16.10 ubuntu drive.
I tried etcher on a HP and a Dell computer, both failed.
Looking forward to a fix.
Rich Prim

jviotti, 2017-04-18:
@lurch @Shou Is any of you running 17.04? I tried to download and install Ubuntu twice, however the installer always fails in both a VM and a NUC.

Shou, 2017-04-18:
@jviotti It seems to work here, both my dev Etcher, and the AppImage from the Etcher website.

richprim, 2017-04-18:
Yes I downloaded it and burned it to a flash drive.
Except for Etcher not working everything else seems to work.
It even got my HP Laserjet P1102w installed without additional cups installation.
Rich Prim

lurch, 2017-04-19:
I've been doing some experimentation in VirtualBox instances running the LiveCD versions of Ubuntu 16.10 and 17.04, and I believe I've tracked down the problem.
In Ubuntu 16.10 and below, the `libgconf` package is a requirement of the `ubuntu-desktop` package, which means that it's always available. However in Ubuntu 17.04 `libgconf` **isn't** required by the `ubuntu-desktop` package, which means that it isn't installed by default. In turn, that means attempting to run Etcher on a default install of Ubuntu 17.04 fails with:
```
/tmp/.mount_mMxWX5/usr/bin/etcher: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory
```
(`gconf` _is_ listed as an Etcher pre-requisite at https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies )
Additionally, in Ubuntu 17.04 `libgconf-2-4` has moved from `Section: libs` to `Section: universe/libs`.

So @richprim to get Etcher working in Ubuntu 17.04 you need to go into the "Software & Updates" application in Ubuntu's "System Settings" section, enable the "Community-maintained free and open-source software (universe)" option, close that dialog and Reload the software index, and then once that's finished open up a Terminal window and type:
```
sudo apt-get install -y libgconf-2-4
```
and then you should finally find that Etcher now works as expected! (but please let us know if it still doesn't work).
Apologies for the inconvenience.

@jviotti This reminds me very much of #835 but I guess unfortunately for us Ubuntu 17.04 is likely to be much more widely used than Fedora 25 :-/

jviotti, 2017-04-19:
@lurch Amazing research! `libgconf-2-4` is already listed as a dependency for the debian package (https://github.com/resin-io/etcher/blob/master/scripts/build/debian/config.json#L23), so I guess this only affects AppImages.

Do you know of any way we can prevent this from happening? Otherwise, would you mind updating the installation instructions to include that one liner that installs the package?

richprim, 2017-04-20:
Thank you for the reply, I will try this in the next few days and get back to you.
Rich Prim

richprim, 2017-04-21:
It works, I already or it was already checked Community-maintenance and open software (universal).
All I did was did the sudo apt-get and then it worked great.
No need for apology, just glad there are people like you writing great software.
Looks like a lot of orginization's are recommending Etcher to burn SD cards.
You are the "go to" application, great work.
Rich Prim

lurch, 2017-04-22:
Thanks for the compliments, and for confirming that Etcher is working for you now :-)

jviotti, 2017-05-09:
Closing this, since there's not much we can do from our side.

lurch, 2017-05-10:
Reopening - I was going to do additional tests (IIRC @jviotti asked me to find out whether this requirement was due to electron or appimages), and I was also going to add some notes to the documentation (for all the users that come across this in future).
Just hadn't got around to either yet I'm afraid.

jviotti, 2017-05-10:
Oh OK, sorry about that!

probonopd, 2017-12-03:
libgconf-2.so.4 needs to be bundled inside the AppImage then.

probonopd, 2018-04-10:
`ubuntu-18.04-beta2-desktop-amd64.iso` is also affected.

probonopd, 2018-05-10:
Workaround: `sudo apt install libgconf2-4`, then the AppImage will run. But it needs to be fixed inside the AppImage.

probonopd, 2018-07-21:
Still

```
me@host:~$ /isodevice/Applications/etcher-electron-1.4.4-x86_64.AppImage 
3226
/tmp/.mount_zrTT19/usr/bin/etcher-electron: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory
```

Please do fix this; the fix is easy and this is getting annoying. Thanks!

jstressman, 2018-09-23:
I concur with @probonopd 

Seeing that this very easy fix has still not been implemented after 19 months is rather irritating.

Is there some reason nobody cares that this program simply does not work on Ubuntu Linux for the past few releases? (Over at least 2 major releases over the course of around a year and a half now.)

It asks if you want to install a desktop file, but then just silently fails after that if you aren't running it from a command line to see the error about libgconf.

closed 2018-10-15 by jviotti

wittend, 2018-10-15:
As of today, Monday October 15, 09:38 CDT, This does not work for me on Ubuntu 18.04.

probonopd, 2018-10-15:
@jviotti did you mean to close this? Imho this should be closed only after libgconf2-4 is bundled inside the AppImage.

wittend, 2018-10-15:
Yes, I agre.  I am unable to reach any repository that will let me install
the required library.  Etcher is evidently useless on the current
mainstream Ubuntu 18.04 distribution.

richprim, 2018-10-16:
I fixed it by opening a terminal and do the following

sudo apt-get install -y libgconf-2-4

Apparently this package is necessary and not installed in 17.04 and above.

Rich Prim

probonopd, 2018-10-16:
> I fixed it by opening a terminal and do the following 
> sudo apt-get install -y libgconf-2-4 

Yes, that is a known workaround, but the solution would be to bundle the library inside the AppImage. An AppImage is supposed to run without the user needing to manually install anything.

richprim, 2018-10-16:
Yes I see, well i didn't know if you were aware of the work around.
Rich Prim

joshirio, 2018-10-17:
Ubuntu 18.04 minimal install is also affected. Why can't it be bundled inside the AppImage?

probonopd, 2018-10-17:
Ubuntu 18.04 Live ISOs are also affected. Which means "default install" is affected, too.

sameer-manek, 2018-10-18:
I see an error box popping up saying "expected a number" when I execute etcher.. anyone getting the same issue?

OS : Ubuntu MATE 18.04.

k3anton, 2018-12-07:
Ubuntu 18.04.1 LTS affected still.

probonopd, 2018-12-31:
Unbelievable it's still not fixed in 1.4.9. It's been almost a year now...

```
me@host:~$ '/home/me/Downloads/balena-etcher-electron-1.4.9-x86_64.AppImage'
2993
/tmp/.mount_G2Gbv3/usr/bin/balena-etcher-electron: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory
```

thundron, 2018-12-31:
We're investigating this, as you can see [here](https://github.com/electron/electron/issues/2727#issuecomment-432700970) though we might not be able to do anything about it, since we already package all the needed dependencies when building the AppImage, *including* the correct libgconf.
We're working on upgrading Electron to solve many other issues so this might be one that will benefit from it as well, given they fix it there.

probonopd, 2018-12-31:
If you need a manual workaround, you can always extract (unpack) the AppImage, add the library, and pack it again. Let me know if you need assistance, I'm happy to walk you through in case the issue can't be solved otherwise

dereks, 2019-02-22:
> we already package all the needed dependencies when building the AppImage, _including_ the correct libgconf

@thundron If that is the case, why does this error still happen? Isn't the whole point of AppImage to include all the dependencies -- including **libgconf** version 2.4?

What is the rationale for blaming this error on electron?

PS> I was directed to Etcher by Hardkernel / Odroid. (Using a fresh Ubuntu 18.04 workstation.)