balena-io/etcher #1311: Setup Electron crash reporter

jviotti, 2017-04-15:
See https://electron.atom.io/docs/api/crash-reporter/.

We should deploy a small server for this: https://github.com/electron/mini-breakpad-server

lurch, 2017-04-20:
Interesting. Could / should this be part of resin-corvus, or is this more low-level than that?

jviotti, 2017-04-20:
Oh, that's a very good idea. @stefan-mihaila what do you think? The tricky part is that this code should be ran on the main process before everything else. resin-corvus could expose a function to configure it that we call there.

lurch, 2018-02-26:
Pinging @izavits in case this is something that relates to resin-analytics?

izavits, 2018-02-28:
We could probably include an integration in the analytics lib for this.

jhermsmeier, 2018-02-28:
@izavits wouldn't that be a bit too specific towards Electron apps?

izavits, 2018-03-01:
@jhermsmeier is it something that can be covered by sentry?

jviotti, 2018-03-01:
This is very Electron specific, and not something that can be covered
with any other JS third party tool given that its part of the C++ side
of things.

closed 2018-10-15 by jviotti