balena-io/etcher #1313: Unmount failed

craigusus, 2017-04-16:
- **Etcher version: Etcher-1.0.0-rc.3-win32-x64.exe**
- **Operating system and architecture: Windows 10 Home 64-bit**

Getting below errors when flashing the latest Rasbian image from [https://downloads.raspberrypi.org/raspbian_latest](https://downloads.raspberrypi.org/raspbian_latest)

[image]

jviotti, 2017-04-17:
Hi @craigusus ,

Thank you very much for reaching out. We're actively trying to fix this issue, however no-one from the Etcher team is able to reproduce, which makes things hard, so any help you can provide us would be incredibly valuable!

Some questions, so I can see if I can reproduce:

- Does this happen *every* single time, or just sometimes? What happens if you retry, with the same image?
- Does this happen if you try to flash another image?
- What does the drive contains *before* flashing to it? Does it have multiple partitions, and therefore many drive letters attached to it?
- What kind of drive is this? Is an SD Card? A USB stick? Are you plugging it in an internal reader?

Can you open DevTools by pressing Ctrl+Alt+I, go to the Console pane, and paste here what you get? There should be some messages prefixed with `[mountutils]` that have unmount specific debugging messages.

Thanks a lot!

68goatconv, 2017-04-17:
Just downloaded the latest version of Etcher. Had the same problem on my WIN 7 64 bit moving an ISO from a SD card onto a boot (empty) 32GB USB.  Everything worked fine until the Unmounting stage.  Never progressed further

jviotti, 2017-04-17:
Thanks a lot for the info. As a workaround, you can go to the settings page (the cog in the top right corner), and disable the eject on success option, until we fix the problem.

craigusus, 2017-04-17:
Hi @jviotti,

Thanks for the the reply...

> - Does this happen every single time, or just sometimes? What happens if you retry, with the same image?

Seems to happen every time, including a retry (log results below)

> - Does this happen if you try to flash another image?

Yes, still fails on another image...
[-1492457692236.txt](https://github.com/resin-io/etcher/files/926487/-1492457692236.txt)

> - What does the drive contains before flashing to it? Does it have multiple partitions, and therefore many drive letters attached to it?

I have tried with empty drives and drives with partitions on already (never had issues on versions before 1.0.0)

> - What kind of drive is this? Is an SD Card? A USB stick? Are you plugging it in an internal reader?

It is a micro SD card, in an adaptor, using an external USB card reader. (again never had issues on versions before 1.0.0)

I have tried with empty drives and drives with partitions on already (never had problems in lesser versions)

DevTools results...
First try:
[-1492456082805.txt](https://github.com/resin-io/etcher/files/926482/-1492456082805.txt)
Retry:
[-1492456647888.txt](https://github.com/resin-io/etcher/files/926483/-1492456647888.txt)

Thanks

fourthhorsem4n, 2017-04-17:
@craigusus just curious, did you have your task manager open whIle using etcher?

craigusus, 2017-04-18:
@fourthhorsem4n I don't recall having it open, no.

jviotti, 2017-04-20:
Hi @craigusus , @68goatconv , @fourthhorsem4n ,

I believe I have a fix for this issue: https://github.com/resin-io-modules/mountutils/pull/26. I created a custom build with the changes that you can download here: https://drive.google.com/open?id=0B7tkbonGU-RyRDVISkl4Sy1QRmc.

Do you mind giving it a go and confirm that the issue is indeed gone?

Thanks a lot!

craigusus, 2017-04-22:
Hi @jviotti,

I have just tried the custom build, and it works with no errors. Other than the Windows format error if I have the 'Eject on success' option ticked:
[image]

Tried with 2 different image files and same results.

jviotti, 2017-04-22:
Thank you very much for the testing. I'm very happy to hear that it worked! I'll release v1.0.0-rc.4 with the fix very soon.

craigusus, 2017-04-22:
You're welcome. Glad to be of assistance.

haberdasher9000, 2017-04-23:
@jviotti 
Thank you for solving this. However, after formatting (I used [this guide](http://www.instructables.com/id/How-to-format-your-SD-card-back-to-the-original-si/) ) and trying again I got the "The write has been completed successfully but Etcher detected potential corruption issues in reading the image back from the drive. It's possible that the drive is corrupted, please consider writing the image to a different one." message. I get my rpi 3 in the mail tomorrow so I'll be able to check if it works anyway.

jviotti, 2017-04-23:
That's great news. I'm releasing this on v1.0.0-rc.4!

closed 2017-04-23 by jviotti

jviotti, 2017-04-24:
Hi @haberdasher9000 ,

> However, after formatting (I used this guide ) and trying again I got the "The write has been completed successfully but Etcher detected potential corruption issues in reading the image back from [...]

That means that Etcher detected that at least one byte was not written correctly. I recommend you to try another SD Card, or another reader!

Zeerob, 2017-04-30:
Hi,
I have just installed v1.0.0-rc.4 (Win64 version), and have used it to flash the hassbian image for a Raspberry Pi 3 to a micro SD card. I get the same Unmount failed error. I have tried it several times, and the same error all the time. However, the flash seems to be successful, it works okay in the Raspberry.

jviotti, 2017-05-01:
Hi @Zeerob ,

Sorry to hear that! Do you mind opening DevTools (with Ctrl+Alt+I), going to the Console pane, and pasting every output you see after the error occurs?

Zeerob, 2017-05-01:
Hi jviotti,
The image seems to work, despite the error. So, I am using it, and have spend quit some time working with it. So, I am reluctant to flash it again, to see the error again, as it would destroy all my work so far. However, the error looked very much like the error that craigusus posted.
Thanks.

jviotti, 2017-05-01:
@Zeerob , the image should indeed have worked, as this is only an optional "eject" step. No worries, if you ever encounter the issue again, take a look a DevTools if you can, since that provides debugging information explaining exactly what when wrong under the hood, in much more detail than the top level error posted in this thread.

Zeerob, 2017-05-01:
Hi jviotti,
Thanks, I will do that!

jviotti, 2017-05-01:
Awesome, thanks a lot! BTW, if the error keeps happening, you can disable automatic ejection after success on the settings screen!