balena-io/etcher #1316: Windows 10 USB devices will not eject if task manager is open.

fourthhorsem4n, 2017-04-17:
- Etcher version: any version
- Operating system and architecture: windows 10 32 and 64 bit versions
- Do you see any meaningful error information on DevTools? N/A

Just something I have noticed with windows 10. The OS will not eject USB devices if the task manager is open giving the message that the device is busy. Closing the task manager then ejecting the USB device solves this problem.

jviotti, 2017-04-18:
Thanks a lot for the valuable observation. I'll try to reproduce this later today.

lurch, 2017-04-20:
How bizarre! I don't have Windows 10 myself, but were you able to reproduce @jviotti ?

jviotti, 2017-04-20:
I tested this out, both from master, and from v1.0.0-rc.3 and I can't seem to be able to reproduce; the drive gets ejected just fine.

Do you mind pasting a screenshot of the error window?

lurch, 2018-02-26:
Unless @fourthhorsem4n is able to provide more info, this should probably be closed as "can't reproduce" ?

jhermsmeier, 2018-02-26:
I've come across this once or twice I think, and thought it was due to the TaskManager being open, but couldn't reliably reproduce it – I think this might very well be classified as a Windows bug, and marked wontfix, if we still can't reproduce it.

jhermsmeier, 2018-04-06:
Duplicate of https://github.com/resin-io/etcher/issues/1370

closed 2018-04-06 by jhermsmeier