balena-io/etcher #1325: YAMLException when drive contains a new line character in its description

jviotti, 2017-04-19:
For example: `description: "foo\nbar"`

This scenario results in:

```
YAMLException: unexpected end of the stream within a double quoted scalar at line 3, column 1:
```

lurch, 2017-04-20:
Is this an Etcher error, or is it drivelist simply not handling YAML correctly?

jviotti, 2017-04-20:
The issue is that if a drive description contains a new line, like `description: "foo\nbar"`, it will be interpreted by the YAML parser as:

```
description: "foo
bar"
```

Which is invalid YAML.

lurch, 2017-04-20:
...in which case surely the problem is that we're giving mal-formed data to the YAML parser, rather than automatically stripping out all newlines? :)

jviotti, 2017-04-20:
Yep, exactly. We can strip down the new lines from the platform scripts before they get to the YAML parser.

lurch, 2017-04-20:
That's not _quite_ what I was suggesting, but never mind... ;-)

jviotti, 2017-04-20:
lol, what is the suggestion?

lurch, 2017-04-20:
Just edited my comment ;)

jviotti, 2017-04-20:
Sounds reasonable. We could play around with YAML multi line string (http://stackoverflow.com/a/21699210), which mainly rely on indentation, by dividing the whole thing by new lines, and checking strings that are not valid YAML, and indenting them.

lurch, 2017-04-21:
We're already using double-quoted strings, so perhaps the fix is to simply escape `\n`, `\t` etc. to `\\n` and `\\t`?  I wonder if https://developer.mozilla.org/en/docs/Web/JavaScript/Reference/Global_Objects/String/raw or https://www.npmjs.com/package/js-string-escape might be of any use here?

lurch, 2018-02-26:
See also https://github.com/resin-io-modules/drivelist/pull/156 and https://github.com/resin-io-modules/drivelist/pull/161

I guess the fact that this hasn't been fixed yet, illustrates exactly how often we come across drive-names that contain embedded newline characters ;-)

jhermsmeier, 2018-04-06:
This is fixed on Windows (by https://github.com/resin-io-modules/drivelist/pull/178) & Mac OS (by https://github.com/resin-io-modules/drivelist/pull/253), and will be fixed on Linux by https://github.com/resin-io-modules/drivelist/pull/255

lurch, 2018-04-07:
It was _already_ fixed on Linux by https://github.com/resin-io-modules/drivelist/pull/156 :wink:

closed 2018-10-15 by jviotti