balena-io/etcher #1340: Hanging on Starting...

Vivalldi, 2017-04-21:
- **Etcher version:1.0.0-rc.3**
- **Operating system and architecture:OSx 10.12.4 64bit**
- **Do you see any meaningful error information on DevTools?**

```
2017-04-10-raspbian-jessie-lite.img","size":{"original":1297862656,"final":{"estimation":false,"value":1297862656}}},"drive":{"device":"/dev/disk3","description":"SD Card Reader","size":7969177600,"mountpoints":[{"path":"/Volumes/Raspbian"}],"raw":"/dev/rdisk3","protected":false,"system":false,"name":"/dev/disk3"},"unmountOnSuccess":true})
WRITER: Attempting to elevate

WRITER: Running: env ELECTRON_RUN_AS_NODE=1 IPC_SERVER_ID=etcher-server-14578 IPC_CLIENT_ID=etcher-client-14578 SKIP=1 '/Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper' /Applications/Etcher.app/Contents/Resources/app.asar/lib/child-writer/writer-proxy.js /Applications/Etcher.app/Contents/Resources/app.asar /REDACTED/2017-04-10-raspbian-jessie-lite.img --drive /dev/disk3 --unmount --check
```

jviotti, 2017-04-24:
Hi @Vivalldi ,

Thank you very much for reporting. That is very weird. We've seen a very similar issue before (https://github.com/resin-io/etcher/issues/985), however looks like that caused by macOS refusing to unmount the drive where in your case, it seems to happen when trying to elevate the process.

Do you get to see the elevation dialog that asks for your password? 

Can you try the following?:

- Open Etcher and try to flash an image, so the issue occurs
- **Without closing Etcher**, copy paste the command that you see on DevTools in a terminal and  prefix it with `sudo`. For example:

```sh
sudo env ELECTRON_RUN_AS_NODE=1 IPC_SERVER_ID=etcher-server-14578 IPC_CLIENT_ID=etcher-client-14578 SKIP=1 '/Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper' /Applications/Etcher.app/Contents/Resources/app.asar/lib/child-writer/writer-proxy.js /Applications/Etcher.app/Contents/Resources/app.asar /REDACTED/2017-04-10-raspbian-jessie-lite.img --drive /dev/disk3 --unmount --check
```

- Run the command
- Do you see any error message?

Vivalldi, 2017-04-25:
I ended up burning by hand on the command line. Will update if I have another SD card to burn later.

lurch, 2017-04-25:
Hahaha, I'm glad to see you didn't mis-type "I ended up burning my hand ..." :laughing:

opethion, 2017-05-09:
I get the same error as OP. Etcher 1.0.0-rc.5. OSX 10.12.4. Tried the command above but no status changes.  Still stuck on "Starting".

clhols, 2017-06-19:
I am seeing the same thing on macOS 10.12.5 with Etcher 1.0.0. I get the elevation dialog but nothing happens after that.

jviotti, 2017-06-21:
@clhols @opethion @Vivalldi I'm really struggling to find the root cause of this issue. Just to check: can you show me the absolute paths of the images you're trying to flash? I'm thinking the location or some weird character in the path might be causing this.

Also, can you show me the permissions of the image and the directory that contains it?

clhols, 2017-06-21:
Mine was:
/Users/clhols/Downloads/androidthings/iot_rpi3.img

jhermsmeier, 2017-11-17:
Fixed in https://github.com/resin-io/etcher/pull/1849

closed 2017-11-17 by jhermsmeier