balena-io/etcher #1341: Release checksums and/or signed releases

imrehg, 2017-04-21:
Sort of a release thing that could go under [etcher-homepage](https://github.com/resin-io/etcher-homepage) as well, but maybe more relevant here: would be great to have checksums displayed somehow for the relevant downloads (`md5` / `sha256` are the most common ones), and/or having the releases signed in a way that it can be verified.

lurch, 2017-04-22:
@jviotti will be able to confirm, but I believe the Windows and Mac builds are already signed?

Also pinging @craig-mulligan as he maintains the https://github.com/resin-io/etcher-homepage

jviotti, 2017-04-24:
Windows and macOS are code-signed, and we plan to do the same for AppImages at some point.

Showing checksums would be a very nice addition :+1:

craig-mulligan, 2017-04-24:
Sounds cool, I've added an issue here: https://github.com/resin-io/etcher-homepage/issues/37

lurch, 2018-01-11:
A `SHASUMS256.txt` is now included on https://github.com/resin-io/etcher/releases for all releases from v1.2.0 onwards, but these checksums aren't displayed on https://etcher.io/ yet.

closed 2018-10-15 by jviotti

snow8gaia, 2019-01-31:
Please, please, please put it on https://etcher.io/ so that its easier for most people. Even a direct link to the [GitHub page](https://github.com/balena-io/etcher/releases) will work. 

I (and some) almost skipped downloading it because of the lack of any checksums.  They help prevent broken downloads on unreliable networks. Happens quite frequently in a lot of places. 

Thanks a lot.

Quantumplation, 2021-05-13:
Is there a reason you stopped providing these checksums?

lurch, 2021-05-13:
The checksums _are_ still available, but they're in a slightly weird format :confused: #2826