balena-io/etcher #1348: Add support for .sdcard extension

sebastient, 2017-04-24:
The .sdcard extension is a common raw image format and supporting it would be handy.  Currently one can rename a .sdcard to .img and it works fine with Etcher but direct support would be even better.

lurch, 2017-04-24:
Interesting, I'd not heard of that before. Can you link us to some examples, that we can use for testing?

sebastient, 2017-04-25:
For example Yocto builds will generate .sdcard and .sdcard.gz images.  They're just normal disk images that would get dd'ed to an /dev/DRIVE on Linux/Mac.

alexandrosm, 2017-04-25:
It's funny you mention that, because resinOS is of course a yocto distro..
so we ourselves make .sdcard files :)

lurch, 2017-04-25:
Ping @agherzan - how come you never told us about this? ;-)

agherzan, 2017-04-26:
These files are not coming from the oe-core but BSPs provide sd card generation mechanisms and most of them use different names for them as there is no standardisation around it. For example TI uses sdimg, RPi uses rpi-sdimg, FSL uses sdcard etc. Even resin uses another extension: resinos-img. I'm not sure about the internal rationale about this filename restriction but if we can't avoid it we could add support for all these variations for the major BSP layers.

closed 2017-06-06 by jviotti

adcurtin, 2018-06-27:
we use .usb, so this is still an issue. The issue should be renamed to reflect what was fixed, or the actual problem should be fixed. IMO the best solution is a warning about unsupported file format, but let me do it anyway. Preventing this because of a file extension just seems wrong.

lurch, 2018-06-28:
@adcurtin see #2106 and #1360

jhermsmeier, 2018-07-02:
@adcurtin yeah, that sounds like a good compromise, we could warn & treat it as a flat octet-stream if continued.