balena-io/etcher #1358: Have Etcher report SD statistics

sund, 2017-04-25:
On the 'Flash Complete' page when also Validating copy, have Etcher show Peak and Average Read and Write Speeds.

Extra Settings:
When 'Validate write on success' is enabled, also show option to write a statistics.txt file to the SD Card with:

Date and time imaged
Elapsed Time to image
Average and Peak Read and Write Speeds.
And on Mac show output of ```diskutil info ``` of the disk for the statistically minded.

lurch, 2017-04-25:
I think we're already planning some of these things for Etcher V2 https://resin.io/blog/the-future-of-etcher/ :-)

But we definitely won't be writing anything to the SD card after flashing, because that would default the whole purpose of Etcher's exact byte-by-byte copying.

sund, 2017-04-25:
rather than save to SD Card, what about to the desktop?

lurch, 2017-04-25:
Depends if saving things "to the desktop" means the same thing for all three platforms we support ;)

We'll bear your request in mind, thank you.

lurch, 2018-02-26:
Connects to #778

closed 2018-10-15 by jviotti