balena-io/etcher #1360: Allow selection of any file type by using a custom filter

jviotti, 2017-04-26:
We should have two extensions filters in the open file dialog:

- Supported images: containing all the images formats we know about
- All Files: which allows the user to select any file

lurch, 2017-04-26:
We believe this should fix #1348

lurch, 2017-05-15:
Haha, I've just realised that this is a duplicate pf #833  ;-)

gromain, 2017-06-01:
It is, but it is definitely needed.

My build system generates `.wic` and `.sdcard`, it would be nice to be able to add these "custom" extensions somehow.

jviotti, 2017-06-01:
@gromain @lurch What do you think about adding these here: https://github.com/resin-io/etcher/blob/master/lib/image-stream/supported.js? We've encountered `.sdcard` multiple times already, so I think that at least that one deserves an entry on that file.

lurch, 2017-06-01:
@jviotti  Yeah, could do, as long as we still plan to also allow a "*.*" filter.
Should `.etch` be removed from that list too, given that we've kind of abandoned that idea in favour of the online catalog?

jviotti, 2017-06-01:
Yeah, I agree with both points!

gromain, 2017-06-02:
Also, wic is the new tool provided by yocto to generate images (akin to .sdcard).
Its default output is a `.wic` file. It would be good to add it too I think.
@lurch, do you take care of the changes, or should I put a pull request?

lurch, 2017-06-02:
We discussed this previously in #1348 - I'd be a bit hesitant to add support for every weird and wonderful file extension that various image-backup/image-creation tools happen to create. What do you think @jviotti ?

> or should I put a pull request?

Sure, that would be very helpful thank you :+1:

gromain, 2017-06-02:
No, I completely understand! I was just stating that wic is becoming more and more used in the yocto community and is slowly replacing `.sdcard` .

lurch, 2017-06-02:
Just out of curiosity, what does `.wic` stand for?

gromain, 2017-06-02:
Wic is a contraction of oeic which stands for Open-Embedded Image Creator.

See chapter 5.9.2.1 [here](https://www.yoctoproject.org/docs/2.3/dev-manual/dev-manual.html#create-wic-images-wic) in the Yocto Manual.

lurch, 2017-06-02:
O, I C
(oh, I see) 
:laughing:

jviotti, 2017-06-02:
> I'd be a bit hesitant to add support for every weird and wonderful file extension that various image-backup/image-creation tools happen to create. What do you think @jviotti ?

Yeah, I agree, let's get the core of this issue (allow any file type selection) done.

jhermsmeier, 2017-06-08:
I know, I'm a bit late to the party – but there's something here that's been sitting in the back of my head: Why do so many tools invent their own filename extensions, even though there are already plenty (with `.img` prob. being the most popular) for the exact same thing?
This seems to cause quite some confusion among users in all corners of the internet – and I was wondering, @gromain, if you might have insights into what drives this?

lurch, 2017-06-08:
Why do so many people invent their own names, even though there are plenty already (with `Jonas` prob. being the most popular) ?
;-D

jhermsmeier, 2017-06-08:
Now that's some quality trolling there, @lurch 😆 
This analogy doesn't really hold up though, unless someone cloned me and named that clone Jonas as well ;P

lurch, 2017-06-08:
With his involvement in the Yocto project, perhaps @agherzan might have some idea why different projects (I think he previously specifically mentioned different BSPs) choose to use different file-extensions for what's essentially the same file-format?

@jhermsmeier Trolling aside, I wonder if it might be as simple as wanting to 'label' which particular tool was used to create a particular disk image? \*shrug\*

gromain, 2017-06-08:
I can only speak of Yocto because of my involvement in using it @jhermsmeier .

But to me, it's exactly as @lurch said, basically you can, in the same image recipe (the script that creates your distribution) generates several different file format, either archive of rootfs, byte-to-byte images of rootfs with different filesystems, complete sdcard images, etc...
It's quite convenient to be able to trace which tool created what, even though I agree, it could be in the name of the file, and not on the extension.

jhermsmeier, 2017-06-12:
@gromain thanks for the explanation! It does make sense to me now, but having it in the filename instead, or as a double extension (so i.e. `imagename.wic.img`) would make things so much easier.

robbawebba, 2018-10-08:
Hi Everyone! I stumbled upon this thread while trying to write a Yocto-generated `.wic` image to a USB device. I usually use the `dd` command from Linux when working with Yocto, but I've used Etcher in the past while working on Windows and Mac and I love it.

I was testing Etcher's `wic` support for my team's electrical engineer who is usually on Windows. I was pretty surprised that files with `wic` extensions aren't technically supported, especially considering ResinOS is created using Yocto. I added a `.img` extension to the end of the file and it works perfectly. 

It's been over a year since this issue was reported and it seems like `wic` has been growing in popularity with Yocto-based Linux distributions. Would you mind if I opened a PR that adds `wic` file extension support?

closed 2018-10-15 by jviotti