balena-io/etcher #1370: failed to unmount

lee-lvd, 2017-04-29:
- **Etcher version:**
- **Operating system and architecture:**
- **Do you see any meaningful error information on DevTools?**

font grey on grey is very hard to read.

I am on Windows 7

1.0.0-rc.4

I am trying to write 3 different SD cards

I tried putting on 
"WeatherStation.img"

I tried putting on
"2017-04-10-raspbian-jessie-lite.img"

I tried putting on 
"2017-04-10-raspbian-jessie.img"

all three failed 
here is the results from the last one I tried
And I am not sure why I can not copy the text to paste here,
I did a window copy using paint to get a copy of the error Unmount failed 
at Error (native)

[image]

jviotti, 2017-05-01:
Hey there,

Thanks for reporting. Can you open DevTools by using Ctrl+Alt+I, going to the Console pane, and pasting everything you see there?

The logging information there will help us see what is really going on.

lurch, 2018-02-26:
Perhaps this should be closed as "can't reproduce", unless @lee-lvd is able to provide more info?

Regarding "font grey on grey is very hard to read." see #1721

jhermsmeier, 2018-02-26:
The unmounts failing on Windows is still a very prevalent issue, which will require some intricate work and testing in [mountutils](https://github.com/resin-io-modules/mountutils) – so I think it'd be good to keep this open, also for people to add info to it

jhermsmeier, 2018-03-09:
Ref duplicates https://github.com/resin-io/etcher/issues/2092 and https://github.com/resin-io/etcher/issues/2110

closed 2018-10-15 by jviotti