balena-io/etcher #1383: Slightly ugly settings screen in 1.0.0-rc.5

lurch, 2017-05-04:
[image]

Just IMHO, but it looks a little "bottom heavy", and I don't like the way the red "Dangerous" box touches the line below it.
Pinging @konmouz and @Shou to do their prettification magic :-)

konmouz, 2017-05-04:
@lurch, settings will be transferred to a modal window and unsafe mode will be moved to the drives' modal. I'll try to find the relevant issues.

lurch, 2017-05-04:
Are those changes still planned to be squeezed in before the release of 1.0.0 ? @jviotti ?

Shou, 2017-05-04:
Maybe we can push it into the empty space on the right? Or remove the 'advanced' header because that's sort of communicated with the 'dangerous' label already?

lurch, 2017-05-04:
Oh, and I keep forgetting to suggest it, but perhaps "Report errors" should be renamed to "Report errors back to resin.io" ? Because otherwise users might not realise what it actually means (I confess when I first saw it, I thought that unticking it meant that any write-errors would be silently ignored).
And this button actually controls both the error-reporting _and_ the statistics-collection, right @stefan-mihaila ?

jviotti, 2017-05-07:
@Shou Let's make a quick fix to solve this, and then we can focus on the new settings screen.

lurch, 2017-05-16:
I've just jun the Fedora 25 LiveCD in a VM (VirtualBox), and after installing `libXScrnSaver` ( #835 ) the Etcher v1.0.0 AppImage runs fine, but the (bottom of the) settings-screen looks even worse than it does on my Ubuntu screenshot above :-(  Perhaps this difference is caused by different Linux distros including slightly different fonts or something?

[image]

closed 2017-05-30 by jviotti