balena-io/etcher #1384: Can't run Etcher appimage on kubuntu 16.10

aseques, 2017-05-04:
- **Etcher version:** 1.0.0-rc5
- **Operating system and architecture:** Kubuntu 16.10, amd64
- **Do you see any meaningful error information on DevTools?**  N/A
I get the error `./Etcher-1.0.0-rc.5-linux-x86.AppImage: error while loading shared libraries: libfuse.so.2: cannot open shared object file: No such file or directory` when I try to run the downloaded image.
After doing a strace I seed that it isn't looking for the place where libfuse2 is installed.
Since ubuntu trusty [libfuse2](http://packages.ubuntu.com/search?suite=trusty&arch=any&mode=exactfilename&searchon=contents&keywords=libfuse.so.2) is installed into either:

```
- /lib/x86_64-linux-gnu/libfuse.so.2
```

aseques, 2017-05-04:
My fault, after doing an strace I saw that I downloaded the x86 version, and I am on a 64bit system. A warning when launching the issue would have been nicer, but still my fault

lurch, 2017-05-04:
Thanks for letting us know :)

closed 2017-05-04 by lurch