balena-io/etcher #1385: SD card not ejected after verification step.

ilovetogetspamed, 2017-05-05:
_- **Etcher version:** 1.0.0-rc.4
- **Operating system and architecture:** iMac17,1running OS X 10.11.6 (15G1421)  
- **Do you see any meaningful error information on DevTools?** No

Nice, simple to use product.

It does not actually eject the SD card that it just created, as it states at the end of the verification process.

It's not a big deal, just wanted to bring it to your attention.  In my case it actually helps my workflow since I need to add/edit files on the SD card image.  Maybe that should be a checkbox option?

Thanks for your work on Etcher.

lurch, 2017-05-05:
There _is_ a checkbox option for it, in the settings screen ;-)  (cog icon in the upper-right)
Please try updating to 1.0.0-rc.5 and let us know if it's still not working as expected.

ilovetogetspamed, 2017-05-05:
Still behaves the same after updating to 1.0.0-rc.5.  Note the 'boot' volume still visible on the desktop above the Etcher window.

I also made sure that 'Auto-unmount on success' was selected.

[image]
[image]

jviotti, 2017-05-07:
That's weird. Maybe there is an edge case on Mountutils that we're not catching? 

@ilovetogetspamed What kind of drive do you have? Can you send us the output of `diskutil info /dev/diskN`?

jviotti, 2017-05-07:
Looks like I'm not able to reproduce this on a MacBook Pro. This is definitely not an issue on mountutils. We need to unmount the drive before being able to flash to it, so we know that its getting unmounted correctly. 

@ilovetogetspamed Can you link us to the exact image you're trying to flash? Could it be that the operating system is re-mounting the drive after we unmount it for some reason (e.g. the image contains certain partition types)?

ilovetogetspamed, 2017-05-08:
diskutil info /dev/disk2
   Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Built In SDXC Reader

   Volume Name:              Not applicable (no file system)

   Mounted:                  Not applicable (no file system)

   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 Secure Digital
   SMART Status:             Not Supported

   Total Size:               8.0 GB (7969177600 Bytes) (exactly 15564800 512-Byte-Units)
   Volume Free Space:        Not applicable (no file system)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Yes
   Media Removal:            Software-Activated

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

OS X El Capitan Version 10.11.6  iMac (Retina 5K, 27-inch, Late 2015) 3.2 GHz Intel Core i5

jviotti, 2017-05-08:
Thanks a lot @ilovetogetspamed ! Do you mind sharing the image that you're flashing as well?

ilovetogetspamed, 2017-05-08:
I get the same results with both images from https://www.raspberrypi.org/downloads/raspbian/.  Raspian Jessie with Pixel and Raspian Jessie Lite.

lurch, 2017-05-08:
Is it possible there's some kind of background service / program on @ilovetogetspamed 's computer that is remounting the freshly-written drive after Etcher has finished flashing it?

ilovetogetspamed, 2017-05-08:
If I manually eject the SD card, it does not get automatically remounted.  I have to physically remove the SD card and reinsert it, for it to mount again.

— Before Reinserting:
$ diskutil info /dev/disk2
Could not find disk: /dev/disk2

— After Reinserting:
$ diskutil info /dev/disk2
   Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Built In SDXC Reader

   Volume Name:              Not applicable (no file system)

   Mounted:                  Not applicable (no file system)

   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 Secure Digital
   SMART Status:             Not Supported

   Total Size:               8.0 GB (7969177600 Bytes) (exactly 15564800 512-Byte-Units)
   Volume Free Space:        Not applicable (no file system)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          Internal
   Removable Media:          Yes
   Media Removal:            Software-Activated

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

jviotti, 2017-05-08:
I can reproduce with the Raspbian images. The unmounting does succeed, but the drive gets mounted back like 1 or 2 seconds afterwards :S Running mountutils again on the drive does the trick, but we have to find out why this is the case.

lurch, 2017-05-08:
Probably be interesting to see if `dd` does the same thing? (Can't check myself, as I don't have a Mac)

ilovetogetspamed, 2017-05-08:
The ‘official (?)’ documentation:  https://www.raspberrypi.org/documentation/installation/installing-images/mac.md

jhermsmeier, 2017-05-09:
@lurch `dd` doesn't do unmounting and such, you have to do that manually, afaik (the raspberry docs seem to reflect that as well)

lurch, 2017-05-09:
Sorry, to clarify I specifically meant the "auto-remounting the image after writing has completed" behaviour ;-)

jhermsmeier, 2017-05-09:
@lurch Oh, well, the OS will attempt to remount it after the handle/descriptor to the unmounted device is closed (can differ between OSs, but Mac OS definitely does that) – and we're basically trying to keep it from doing that

closed 2017-05-12 by jviotti

ilovetogetspamed, 2017-05-12:
> Closed #1385 <https://github.com/resin-io/etcher/issues/1385> via #1416 <https://github.com/resin-io/etcher/pull/1416>.

Thanks for your quick work!