balena-io/etcher #1386: Windows: allows drag and drop of .img file on .exe

NicSil, 2017-05-05:
Detail but VERY important when you have many .img to flash.
Choosing the source from the file chooser is boring.

lurch, 2017-05-05:
Just to clarify - do you mean drag & drop a .img file into the application window (which I believe should work already?), or drag & drop the .img onto the .exe application icon on the desktop? (which I guess involves [this](http://stackoverflow.com/questions/464922/how-can-i-drag-a-file-to-an-exe-i-create-drag-and-drop-over-the-icon-and-have) )

MrDiamond123, 2017-05-10:
I think he means drag to the icon on the desktop

I

NicSil, 2017-05-10:
Yes, drag on the desktop icon (which is an alias in the .img folder to the .exe) it works with Win32DiskImager and avoids having to select the files from the app which is always a pain. just drag the .img and its ready to flash

NicSil, 2017-09-01:
Still waiting about this feature: clicking on "select image" and browsing the disk is much more difficult than dragndroping a file on the icon.

jviotti, 2017-09-01:
@NicSil I don't think anyone is working on it yet, but feel free to attempt a contribution yourself.

NicSil, 2017-09-01:
Ok. As i remember my previous win32 C++ dev, bout 20 years ago, taking into account the dropped filename is about 5mn of work ...

lurch, 2017-09-01:
@jviotti According to that StackOverflow thread I linked to, the only requirement is to have the GUI accept the image's filename as the single command-line argument. But I've no idea if Electron would 'get in the way' with adding that.

wrboyce, 2017-09-18:
It would be great if I could do this on macOS too (drop an image onto the Etcher Dock icon).

closed 2018-10-15 by jviotti

NicSil, 2018-10-15:
great.