balena-io/etcher #1389: Detect the protocol of devices

jviotti, 2017-05-07:
For example: PCI or USB.

We should send this to Mixpanel.

lurch, 2017-05-08:
AFAIK this is easy on Linux - if the raw device is `/dev/mmcblkX` then it's a PCI-connected SD reader (or GPIO-connected in the case of the Raspberry Pi's built-in microSD slot), but if it's `/dev/sdX` then it's USB (or SATA) connected, and if it's `/dev/hdX` then it's IDE-connected. Are we able to look at the existing statistics to see if any other device-name patterns show up (under Linux)?
On some laptops, the internal SD card slot is connected via PCI, and on other laptops the internal card slot is connected via USB. (obviously any external SD readers will always be connected via USB!)

I've got no idea yet how easy / hard this detection would be for other OSes ;-)  Maybe this would be a feature worth waiting until we rewrite drivelist as a native C++ addon? (or maybe it's already easy to add to the shell scripts we're already using)

jviotti, 2017-05-08:
The protocol is directly exposed by Diskutil on macOS, so that's trivial. I'd have to research on the Windows side of things.

lurch, 2018-02-26:
@jhermsmeier did a bunch of work on the Windows-side of this in https://github.com/resin-io-modules/drivelist/pull/243

jhermsmeier, 2018-03-23:
Bus types / protocols now also supported on Mac OS since resin-io-modules/drivelist#253

closed 2018-10-15 by jviotti