balena-io/etcher #1395: Windows: Command Failed at startup when the username contains ampersands

jviotti, 2017-05-09:
This happens as soon as you open the app:

```sh
Command failed: C:\Users\Foo&Bar\AppData\Local\Temp\9DB4.tmp.bat 'C:\Users\Foo' is not recognized as an internal or external command, operable program or batch file. The system cannot find the path specified. (code 1
```

This can be more easily reproduced by placing `drivelist` in a directory containing an ampersand and executing the example file:

```sh
git clone https://github.com/resin-io-modules/drivelist
move drivelist "Foo & Bar"
cd "Foo & Bar"
node example.js
```

This file is executed by `child_process.execFile()`. Simple tricks like quoting the whole path don't work. I also tried escaping the ampersand with `^&`.

I did make it work by setting the `execFile` `cwd` option to the directory name of the script and then simply executing the basename, however that won't play well with `asar` archives. And to put another constraint to this: `.execFile` is the only spawning function we can use, since its the only one that is patched by Electron.

lurch, 2017-05-09:
Blimey, sounds fairly nasty. Do we know if this has affected any users in the wild? I guess the nature of this bug (and the related ones) means that Etcher would fail before it gets the chance to log any errors?

lurch, 2017-05-09:
FWIW, if I try replicating your example above on Linux (in `/tmp`), it works okay as long as I do `npm install` after doing `cd "Foo & Bar"`. These all work:
```bash
node example.js
cd ..
node ./Foo\ \&\ Bar/example.js
node Foo\ \&\ Bar/example.js
node "Foo & Bar"/example.js
node "Foo & Bar/example.js"
cd ~
node /tmp/Foo\ \&\ Bar/example.js
node "/tmp/Foo & Bar/example.js"
node /tmp/"Foo & Bar"/example.js
```
and as we're not using any 'special' characters or variables here, these all work identically with double or single cream... I mean quotes! :wink:

jviotti, 2017-05-10:
> Blimey, sounds fairly nasty. Do we know if this has affected any users in the wild? I guess the nature of this bug (and the related ones) means that Etcher would fail before it gets the chance to log any errors?

Yeah, I discovered this one through Sentry :/ The error happens after the app is started, so Corvus does get a chance to report it.

lurch, 2017-05-10:
I've been attempting to dig into this some more, using `drivelist` as a nice **&** small test case...
(in the hope that this might also help with #1376 )

On Linux, it looks like I can use any crazy characters in the pathname that I like, and everything "just works" :
```bash
cd /tmp/Foo\ \!\"£\$%\^\&\*\(\)\'\\\:\;@#~\`\|\\\[\]\{\}\?\<\>-_\=+\$\{PATH\}\ Bar/
$ node example.js 
[
  {
    "device": "/dev/sda",
    "description": "SanDisk SD8SN8U1",
    "size": 1024209543168,
    "mountpoints": [
      {
        "path": "/"
      },
      {
        "path": "/home"
      },
      {
        "path": "/boot/efi"
      },
      {
        "path": "/var/lib/docker/aufs"
      }
    ],
    "raw": "/dev/sda",
    "protected": false,
    "system": true
  }
]
```
Without all the crazy backslash-escapes auto-created by tab-completion, the actual directory name looks like
```
/tmp/Foo !"£$%^&*()'\:;@#~`|\[]{}?<>-_=+${PATH} Bar
```
and from my home-dir, I can run any of:
```bash
node '/tmp/Foo !"£$%^&*()'\''\:;@#~`|\[]{}?<>-_=+${PATH} Bar/example.js'
node /tmp/Foo\ \!\"£\$%\^\&\*\(\)\'\\\:\;@#~\`\|\\\[\]\{\}\?\<\>-_\=+\$\{PATH\}\ Bar/example.js
node "/tmp/Foo "\!"\"£\$%^&*()'\\:;@#~\`|\\[]{}?<>-_=+\${PATH} Bar/example.js"
node "/tmp/Foo "'!'"\"£\$%^&*()'\\:;@#~\`|\\[]{}?<>-_=+\${PATH} Bar/example.js"
```
with identical results. (apparently the `!` causes problems (history expansion) in double-quoted strings in an interactive terminal, but [not in shell scripts](https://unix.stackexchange.com/a/209723) )

On Windows, these are the illegal filename-characters: `\ / : * ? " < > |`, and here's what I found from my investigations:

| Folder | node example.js | node _folder_/example.js | _folder_/win32.bat |
|---|---|---|---|
| | | | |

(blargh, 3am now so I'll finish these notes off tomorrow...)

Possibly relevant / useful: https://github.com/sindresorhus/opn/pull/47

jviotti, 2017-05-10:
Thanks a lot for helping me out with this.

> I'm beginning to wonder if using cmd /c ... with .execFile might be a workaround?)

The problem is still `asar` archives. The only way to execute a file inside an `asar` is to pass its path as the *first* argument to `.execFile` (since Electron understands that, and will handle extraction). Any other variations means that the file will not execute.

***

There is a solution though, which will actually make my work on the CLI way easier: inline the drivelist scripts in JavaScript files that manually extract them out and execute them.

The logic put there can execute the file using my `cwd` workaround, and then it all should work. What do you think?

lurch, 2017-05-10:
> The problem is still asar archives. The only way to execute a file inside an asar is to pass its path as the first argument to .execFile (since Electron understands that, and will handle extraction). Any other variations means that the file will not execute.

Ahhh, thanks. I didn't realise that the `.execFile` restriction was due to the way that Etcher dynamically unpacks things from the asar.

> There is a solution though, which will actually make my work on the CLI way easier: inline the drivelist scripts in JavaScript files that manually extract them out and execute them.

Huh? That sounds hideous :-/ But I guess it _might_ work, as long as we don't run into permissions / threading / forking issues.
However I wouldn't want to see the drivelist scripts 'manually' embedded inside other Javascript files, as that would make them totally unmaintainable. We should continue to edit the platform-specific scripts as standalone files, and then have some step of the build-process that then embeds them inside another JS file (e.g. as base64 or something).

jviotti, 2017-05-10:
Check https://github.com/resin-io-modules/drivelist/pull/164. I created a small script that creates a JSON file out of them, so we can still keep separate for developing purposes. I'll test this thoroughly before merging.

closed 2017-05-10 by jviotti

lurch, 2017-05-11:
Remind me to look into this again sometime next week ;-)

pizzapants, 2017-05-13:
Not sure if this is the same or related error but didnt want to open up another issue if it was. Getting this on startup. The wierd thing is, I havent updated Ethcer recently that Im aware of but it started doing it all of a sudden. Uninstalled and reinstalled the latest from the site did not resolve. 

[Picture of error](http://imgur.com/a/gYMPB)

lurch, 2017-05-15:
@pizzapants That indeed looks like a different problem. Could you please open a new issue, including as much information as possible? Thanks.