balena-io/etcher #1397: Corvus emitting `[object Object]` error to Sentry

jviotti, 2017-05-09:


lurch, 2017-05-10:
Hmm, I guess this never got tracked down then?

jviotti, 2017-05-10:
It got greatly reduced (like by 3/4), but there are still some occurrences that we should track down.

Shou, 2017-06-30:
Looks like `lib/child-writer` uses a bit of `.toString()` which may be the cause.

jviotti, 2017-06-30:
See the conversation in Flowdock. Its corvus applying `JSON.stringify()` on error objects, which results in empty objects, and also the `utils.hideAbsolutePathsInObject` function

jviotti, 2017-06-30:
Looks like an easy fix though, for which we can even have some nice unit tests. Lets fix whatever `utils.hideAbsolutePathsInObject` is doing wrong and create and test a function that is able to stringify errors and plain objects.

jviotti, 2017-06-30:
Check this SO answer: https://stackoverflow.com/a/18391400/1641422. We can use a replacer function for `JSON.stringify()`:

```js
function replaceErrors(key, value) {
    if (value instanceof Error) {
        var error = {};

        Object.getOwnPropertyNames(value).forEach(function (key) {
            error[key] = value[key];
        });

        return error;
    }

    return value;
}

var error = new Error('testing');
error.detail = 'foo bar';

console.log(JSON.stringify(error, replaceErrors));
```

jviotti, 2017-06-30:
Looks like no lodash function works with Error object. `utils.hideAbsolutePathsInObject` fails at a `_.deepMapValues` call, which doesn't really map anything, and thus returns an empty object.

jviotti, 2017-06-30:
I guess an easy way to do it would be to create a function that converts an Error object to a plain object, but then `utils.hideAbsolutePathsInObject` should output an Error instance if that's what came in, so we would need to have a "reconstruct" function.

Probably easier to re-implement `deepMapValues` so that it can map through error objects correctly.

Shou, 2017-07-04:
Wouldn't `JSON.stringify` still return `{}` if we gave it an `Error` object, so shouldn't we return an object from `hideAbsolutePathsInObject`? Or just report the `Error` directly and skip `hideAbsolutePathsInObject`?

lurch, 2017-07-04:
> Or just report the Error directly and skip hideAbsolutePathsInObject?

Is it possible that some of the error details (description) might contain the absolute paths that we're trying to hide?

Shou, 2017-07-04:
Ah, so maybe the idea is to use the same function for both, but only pure objects get turned into a string with `JSON.stringify` before sent to the console.

jviotti, 2017-07-04:
@Shou This problematic logic only applies when logging the error in DevTools, independently on the error being sent to Sentry. We stringify them since its very hard to view objects on DevTools otherwise. Making `hideAbsolutePathsInObject` work on Error objects and return a plain object should do the trick.

Shou, 2017-07-05:
I've made https://github.com/resin-io-modules/resin-corvus/pull/22 now so let me know what you think.

jviotti, 2017-07-17:
I can't tell for sure, given this issue has been particularly difficult to track, but as far as my experiments went (that involved sending actual errors to Sentry and check them out in the dashboard), this issue seems fixed by https://github.com/resin-io/etcher/pull/1604

jviotti, 2017-07-18:
This issue was probably closed by https://github.com/resin-io/etcher/pull/1604. Let's keep an eye on the analytics and re-open if it happens again.

closed 2017-07-18 by jviotti

jorgemanrubia, 2017-11-03:
In case it helps, I am still seeing these `[Object object]` errors. In my case it's an ember app.

[image]

Thanks

lurch, 2017-11-07:
@jorgemanrubia I'm confused - Etcher isn't an ember app, so why have you posted a comment on an Etcher issue? Could you add more details?

jorgemanrubia, 2017-11-07:
Ugh sorry about that @lurch, I meant to post this in the sentry issue tracker. I'll just delete my comment