balena-io/etcher #1410: no instructions how to install etcher found on the website

laplasz, 2017-05-11:
- **Etcher version: 1.0.0-rc5**
- **Operating system and architecture: Linux x64**

Once I downloaded the zip package for Linux x64 I found the following file:
Etcher-1.0.0-rc.5-linux-x64.AppImage

What should I do with this? Any instructions how to deal with .AppImage file? or what i did it wrong?
Thanks, Balazs

laplasz, 2017-05-11:
Oke, I found something:
https://askubuntu.com/questions/774490/what-is-an-appimage-how-do-i-install-it
For me it was not so obvious right now
Thanks

lurch, 2017-05-11:
Just to add to those instructions - we specifically distribute our AppImages in a zip file, so that the executable bit is preserved and you don't need to manually set it after extracting the AppImage from the zip ;-)  I.e. it should just be a case of extract the AppImage from the zip file (e.g. to your home directory or desktop), and you're then immediately able to double-click on the AppImage to launch Etcher.

See also https://github.com/resin-io/etcher-homepage/issues/26

jviotti, 2017-05-11:
I think this can be closed now. Please re-open if you have any other questions :)

closed 2017-05-11 by jviotti