balena-io/etcher #1412: flashing process got stuck at 94% - no errors

laplasz, 2017-05-11:
- **Etcher version: 1.0.0-rc5**
- **Operating system and architecture: Linux x64**

First of all I had to execute the Etcher with sudo to able to start flasing my SD Card. Without root permission I got this

    Error: Command failed: /usr/bin/sudo -n -E -- env ELECTRON_RUN_AS_NODE=1 IPC_SERVER_ID=etcher-server-18480 IPC_CLIENT_ID=etcher-client-18480 SKIP=1 /home/ebalbar/Downloads/Etcher-1.0.0-rc.5-linux-x64.AppImage bin/resources/app.asar/lib/child-writer/writer-proxy.js bin/resources/app.asar /home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip --drive /dev/mmcblk0 --unmount --check
    pmplugin6.0.0 (040):3343 Cannot prompt the user for input when running in non-interactive mode (-n) 1516.3 PAM failed to authenticate [Conversation error]

    at ChildProcess.exithandler (child_process.js:223:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

After start flashing It seamed that got stuck in the beginning of the flashing since only info was available after pressing the Flash! button is the Starting... text on the button for several minutes. Then suddenly got a 94% progress.. and that is it. Nothing after that for an hour. No errors.
By the way I wanted to flash a raspberry pi jessie lite - they recommended this tool..

lurch, 2017-05-11:
The 'sudo -n' error is weird. What exact Linux distro (and desktop) are you using? Does it have a polkit agent installed?
https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md

Can you try the process again please, and copy'n'paste the contents of the DevTools console here? (press `Ctrl + Alt + I` at the same time)

laplasz, 2017-05-11:
it is a custom distro based on Ubuntu 14.04 desktop is LXDE.
i have these package-s containing polkit:
ii  gir1.2-polkit-1.0                                     0.105-4ubuntu3.14.04.1                                         amd64        GObject introspection data for PolicyKit
ii  libpolkit-agent-1-0:amd64                             0.105-4ubuntu3.14.04.1                                         amd64        PolicyKit Authentication Agent API
ii  libpolkit-backend-1-0:amd64                           0.105-4ubuntu3.14.04.1                                         amd64        PolicyKit backend API
ii  libpolkit-gobject-1-0:amd64                           0.105-4ubuntu3.14.04.1                                         amd64        PolicyKit Authorization API
```
 _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:36:20 GMT+0200 (CEST) Application start
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:36:20 GMT+0200 (CEST) Checking for updates ({"currentVersion":"1.0.0-rc.5","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":true})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:36:21 GMT+0200 (CEST) Update notification skipped ({"reason":"Latest version"})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:36:56 GMT+0200 (CEST) Open image selector
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:37:08 GMT+0200 (CEST) Select image ({"path":"/home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip","size":{"original":1297862656,"final":{"estimation":false,"value":1297862656}},"extension":"img","archiveExtension":"zip","logo":false,"bmap":false})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:37:23 GMT+0200 (CEST) Flash ({"image":"/home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip","drive":{"device":"/dev/mmcblk0","description":"SU16G","size":15931539456,"mountpoints":[{"path":"/media/ebalbar/boot"},{"path":"/media/ebalbar/f2100b2f-ed84-4647-b5ae-089280112716"}],"raw":"/dev/mmcblk0","protected":false,"system":false,"name":"/dev/mmcblk0"},"uuid":"f26a9246-121d-4340-9890-230744ec5d07","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Attempting to elevate

/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Running: /home/ebalbar/Downloads/Etcher-1.0.0-rc.5-linux-x64.AppImage bin/resources/app.asar/lib/child-writer/writer-proxy.js bin/resources/app.asar /home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip --drive /dev/mmcblk0 --unmount --check

/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:37:25 GMT+0200 (CEST) Flash error ({"error":{"code":1},"image":"/home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip","drive":{"device":"/dev/mmcblk0","description":"SU16G","size":15931539456,"mountpoints":[{"path":"/media/ebalbar/boot"},{"path":"/media/ebalbar/f2100b2f-ed84-4647-b5ae-089280112716"}],"raw":"/dev/mmcblk0","protected":false,"system":false,"name":"/dev/mmcblk0"},"uuid":"f26a9246-121d-4340-9890-230744ec5d07","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:37:25 GMT+0200 (CEST) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
/tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 {"code":1}
console.(anonymous function) @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51
logException @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:226
reportException @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/lib/gui/modules/error.js:50
ImageWriterService.flash.then.catch @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/lib/gui/pages/main/controllers/flash.js:84
processQueue @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:16843
(anonymous) @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:16887
$digest @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:17982
(anonymous) @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:18211
completeOutstandingRequest @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:6274
(anonymous) @ /tmp/.mount_wlaifX/usr/bin/resources/app.asar/node_modules/angular/angular.js:6554
```

laplasz, 2017-05-11:
with sudo:

```
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:57:30 GMT+0200 (CEST) Application start
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:57:30 GMT+0200 (CEST) Checking for updates ({"currentVersion":"1.0.0-rc.5","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":true})
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:57:31 GMT+0200 (CEST) Update notification skipped ({"reason":"Latest version"})
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:57:40 GMT+0200 (CEST) Open image selector
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:57:52 GMT+0200 (CEST) Select image ({"path":"/home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip","size":{"original":1297862656,"final":{"estimation":false,"value":1297862656}},"extension":"img","archiveExtension":"zip","logo":false,"bmap":false})
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 14:58:08 GMT+0200 (CEST) Flash ({"image":"/home/ebalbar/Downloads/2017-04-10-raspbian-jessie-lite.zip","drive":{"device":"/dev/mmcblk0","description":"SU16G","size":15931539456,"mountpoints":[],"raw":"/dev/mmcblk0","protected":false,"system":false,"name":"/dev/mmcblk0"},"uuid":"38aa3041-1568-4dd5-8b80-a65572207508","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Re-spawning with elevation

/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Reading /proc/mounts
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: rootfs
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: sysfs
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: proc
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: udev
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: devpts
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/dm-1
8/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: none
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sda1
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: binfmt_misc
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: systemd
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: //vhub.rnd.ki.sw.ericsson.se/home/ebalbar
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: gvfsd-fuse
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: Etcher-1.0.0-rc.5-linux-x64.AppImage
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Closing /proc/mounts
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 0% at 2.38 MB/s (eta 543s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 0% at 4.08 MB/s (eta 315s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 1% at 5.72 MB/s (eta 224s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 1% at 6.32 MB/s (eta 201s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 2% at 7.02 MB/s (eta 180s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 2% at 7.07 MB/s (eta 179s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 3% at 7.83 MB/s (eta 160s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 3% at 7.93 MB/s (eta 158s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:15 GMT+0200 (CEST) Progress (write): 3% at 7.76 MB/s (eta 161s)
[... cut ...]
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:17 GMT+0200 (CEST) Progress (write): 90% at 7.3 MB/s (eta 16s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:17 GMT+0200 (CEST) Progress (write): 91% at 7.32 MB/s (eta 15s)
/tmp/.mount_ZtsUNT/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu May 11 2017 15:01:17 GMT+0200 (CEST) Progress (write): 91% at 7.31 MB/s (eta 15s)
```

lurch, 2017-05-11:
Thanks for the logs. In the first log, it looks like there _is_ an error, it's just not being displayed properly? Ping @jviotti 
In the second log, it's very strange that it just hangs at 91%. Given that it 'froze' at 94% the first time, but at 91% the second time, that might point to a faulty SD card or a faulty SD reader? If you have alternative cards and/or readers you could try for comparison, that might be very helpful.

And the image probably still booted okay, because the last 9% of the image might just be unused space? Or more critically, it might contain files that aren't needed to boot, but would actually be corrupt if they're needed e.g. by a particular application?

jviotti, 2017-05-15:
Hm, this is a weird one. @laplasz is there a way we can get our hands on that custom Linux distro? If not, do you mind sending us the `sudo` man page that you have there?

Can you try unzipping the Raspbian image and flashing it that way (so we can validate if the error comes from the unzipping routine)?

jhermsmeier, 2017-11-17:
This should be fixed by https://github.com/resin-io/etcher/pull/1843

closed 2017-11-17 by jhermsmeier