balena-io/etcher #1420: Enable 'Unsafe Mode' in CLI

Stuka87, 2017-05-12:
- **Etcher version:** 1.0
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** ext4 image written to mSATA drive via USB enclosure
- **Do you see any meaningful error information in the DevTools?** As I am using the CLI, the DevTools are not an option. But I can say that since the mSATA drive is not considered a removable device inside Windows, it does not show up as an available drive in the CLI. I can write to it fine via the GUI if I enable Unsafe Mode.

The ability to pass in a flag to enable Unsafe mode would resolve this issue. The GUI does not fit into our workflow, which is why we would like to see this feature in the CLI.

Thanks!

jviotti, 2017-05-15:
Hi @Stuka87,

Thanks a lot for reaching out. Looks like the interactive drive selector widget doesn't allow us (we'll fix it soon!), but you should be able to use `--device` just fine.

Let us know how it goes!

lurch, 2017-05-15:
I guess if the device passed with `--drive` is detected as a system device, maybe we should also ask the user for a confirmation (after displaying a bit of info about the drive), unless the `--yes` flag was passed?

Also, @Stuka87 suggested in gitter that it'd be useful for the examples output by `--help` to be tailored to the OS on which the Etcher CLI is actually running.

Stuka87, 2017-05-15:
As an update, specifying the drive directly does work with the CLI in its current form. Although it should be noted etcher.exe must be run from cmd. If it is run from a PowerShell prompt it will error out, this is mostly likely due to an encoding difference with PowerShell and the drive path that gets passed in. I did not spend too much time digging into a fix, as running from cmd is fine for this use case.

For the time being, I am using the standalone drivelist node module to grab the current list of drives, and then parse that to find the drive I want to write the image too.

gbnyc26, 2018-01-22:
As an update to https://github.com/resin-io/etcher/issues/1966 on MacOS.  Executing the following seems to work:

sudo ./etcher -d /dev/disk2 file.img

In my original issue, I was using -d /dev/disk2s1.  It seems that the device name /dev/disk2s1 is not recognized.

lurch, 2018-01-22:
Yeah, Etcher needs to have the **disk** name (in this case `/dev/disk2`), whereas `/dev/disk2s1` is the name of a **partition** on the disk.

gbnyc26, 2018-01-23:
Thanks for confirming... perhaps documentation should mention this... as it is all too easy to run mount and see where your SD card is mounted and assume that the /dev name is what is needed

closed 2018-10-15 by jviotti