balena-io/etcher #1423: Flashing Stuck at 40%

RyanTWT, 2017-05-13:
- **Etcher version: 1.0.0**
- **Operating system and architecture: MacOS Sierra 10.12.4**
- **Image flashed: retropie-4.2-rpi2_rpi3.img.gz**
- **Do you see any meaningful error information in the DevTools?: Not sure what I should be really looking for.**

[image]

lurch, 2017-05-15:
That error message would seem to indicate that the file you're trying to flash (retropie-4.2-rpi2_rpi3.img.gz) hasn't been downloaded properly. Can you check that the file has a size of 631,583,123 bytes before you try flashing it with Etcher?
In fact, I've just tried writing `retropie-4.2-rpi2_rpi3.img.gz` (which I downloaded from https://github.com/RetroPie/RetroPie-Setup/releases ) with Etcher myself (it worked), and Etcher lists the image size as `2.2 GB` whereas your screenshot shows it as `4.1 GB`, which would also seem to indicate a corrupted download?

jviotti, 2017-05-15:
Hi @RyanTWT ,

Would you mind uploading the *exact* file you're flashing somewhere we can check?

lurch, 2017-05-15:
@jviotti @jhermsmeier Is this another example of where we should look for specific error messages using a regex (like in #1369 ) and then convert them to "friendlier" error messages?

alexandrosm, 2017-05-15:
yeah, this looks like we should tell the user "it appears your source file
is corrupted. are you sure it was downloaded correctly?"

lurch, 2018-02-26:
@RyanTWT Do you still get this problem in the latest Etcher 1.3.1 ?

jhermsmeier, 2018-02-26:
I think we can close this for now – @RyanTWT please feel free to re-open if this issue persists in the latest version.

closed 2018-02-26 by jhermsmeier